phanrahan/magma #415: CoreIR Simulator Crashes When Trying To Read Value From Tuple

David-Durst, 2019-06-23:
The test https://github.com/David-Durst/aetherling/blob/master/tests/test_tuple.py#L65-L90 fails on line 90 with a CoreIR error.

The test creates a tuple from an Array[8, Bit] and a Bit. CoreIR crashes when I try to read the Bit value out.

```
(aetherling) durst@DN800cae09:~/dev/W17-8/aetherling$ pytest -s tests/test_tuple.py::test_atomTupleAppender
/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
================================================ test session starts =================================================
platform darwin -- Python 3.7.1, pytest-4.4.1, py-1.8.0, pluggy-0.9.0
rootdir: /Users/durst/dev/W17-8/aetherling
plugins: xdist-1.28.0, forked-1.0.2
collected 1 item

tests/test_tuple.py /Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Starting topological sort
topo_order.size() = 4
numVertices(g)    = 4
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file /Users/durst/dev/W17-8/coreir/2coreir/src/../include/coreir/ir/casting/casting.h, line 249.
Fatal Python error: Aborted

Current thread 0x000000010bd815c0 (most recent call first):
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/coreir/simulator.py", line 42 in get_value
  File "/Users/durst/dev/W17-8/magma/magma/simulator/coreir_simulator.py", line 208 in get_value
  File "/Users/durst/dev/W17-8/aetherling/tests/test_tuple.py", line 90 in test_atomTupleAppender
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/python.py", line 178 in pytest_pyfunc_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/python.py", line 1464 in runtest
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 123 in pytest_runtest_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 198 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 226 in from_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 198 in call_runtest_hook
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 173 in call_and_report
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 93 in runtestprotocol
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 78 in pytest_runtest_protocol
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 270 in pytest_runtestloop
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 249 in _main
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 209 in wrap_session
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 242 in pytest_cmdline_main
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/config/__init__.py", line 79 in main
  File "/Users/durst/anaconda/envs/aetherling/bin/pytest", line 11 in <module>
Abort trap: 6
```

leonardt, 2019-06-25:
Did a small modification to run the test under lldb and get a backtrace, here's what I got
```
❯ git diff
diff --git a/tests/test_tuple.py b/tests/test_tuple.py
index 34ca2c0..d074fe4 100644
--- a/tests/test_tuple.py
+++ b/tests/test_tuple.py
@@ -89,3 +89,6 @@ def test_atomTupleAppender():
     assert seq2int(sim.get_value(testcircuit.O[0], scope)) == 4
     print(sim.get_value(testcircuit.O[1], scope))
     #assert sim.get_value(testcircuit.O[1], scope) == True
+
+
+test_atomTupleAppender()
```
```
~/repos/aetherling master* 8s
❯ lldb --python tests/test_tuple.py
/Library/Developer/CommandLineTools/Library/PrivateFrameworks/LLDB.framework/Resources/Python

~/repos/aetherling master*
❯ lldb -- python tests/test_tuple.py
(lldb) target create "python"
Current executable set to 'python' (x86_64).
(lldb) settings set -- target.run-args  "tests/test_tuple.py"
(lldb) r
Process 6372 launched: '/Users/lenny/miniconda3/bin/python' (x86_64)
/Users/lenny/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
/Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Starting topological sort
topo_order.size() = 4
numVertices(g)    = 4
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file /Users/lenny/repos/coreir/src/../include/coreir/ir/casting/casting.h, line 249.
Process 6372 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff7dfd22c6 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff7dfd22c6 <+10>: jae    0x7fff7dfd22d0            ; <+20>
    0x7fff7dfd22c8 <+12>: movq   %rax, %rdi
    0x7fff7dfd22cb <+15>: jmp    0x7fff7dfcc457            ; cerror_nocancel
    0x7fff7dfd22d0 <+20>: retq
Target 0: (python) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff7dfd22c6 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff7e08dbf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff7df3c6a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff7df0520d libsystem_c.dylib`__assert_rtn + 324
    frame #4: 0x0000000105c4c254 libcoreir.dylib`CoreIR::cast_retty<CoreIR::Select, CoreIR::Wireable*>::ret_type CoreIR::cast<CoreIR::Select, CoreIR::Wireable>(CoreIR::Wireable*) + 100
    frame #5: 0x000000010678c9b5 libcoreirsim.dylib`CoreIR::toSelect(CoreIR::Wireable*) + 21
    frame #6: 0x000000010678c5b8 libcoreirsim.dylib`CoreIR::SimulatorState::getValue(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 88
    frame #7: 0x0000000106797f8b libcoreirsim.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 59
    frame #8: 0x00000001067986ab libcoreirsim.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 1883
    frame #9: 0x00000001067986ab libcoreirsim.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 1883
    frame #10: 0x0000000106797f11 libcoreirsim.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) + 65
    frame #11: 0x0000000106731191 libcoreirsim-c.dylib`CORESimGetValueByOriginalName + 97
    frame #12: 0x000000010355f884 libffi.6.dylib`ffi_call_unix64 + 76
    frame #13: 0x000000010355ee8b libffi.6.dylib`ffi_call + 939
    frame #14: 0x0000000103543c58 _ctypes.cpython-37m-darwin.so`_ctypes_callproc + 632
    frame #15: 0x000000010353decc _ctypes.cpython-37m-darwin.so`PyCFuncPtr_call + 268
    frame #16: 0x000000010002f283 python`_PyObject_FastCallKeywords + 179
    frame #17: 0x000000010016cc35 python`call_function + 453
    frame #18: 0x000000010016a852 python`_PyEval_EvalFrameDefault + 46114
    frame #19: 0x000000010015e1fe python`_PyEval_EvalCodeWithName + 414
    frame #20: 0x000000010002f783 python`_PyFunction_FastCallKeywords + 195
    frame #21: 0x000000010016cb27 python`call_function + 183
    frame #22: 0x000000010016a7bc python`_PyEval_EvalFrameDefault + 45964
    frame #23: 0x000000010015e1fe python`_PyEval_EvalCodeWithName + 414
    frame #24: 0x000000010002f783 python`_PyFunction_FastCallKeywords + 195
    frame #25: 0x000000010016cb27 python`call_function + 183
    frame #26: 0x000000010016a7bc python`_PyEval_EvalFrameDefault + 45964
    frame #27: 0x000000010002f075 python`function_code_fastcall + 117
    frame #28: 0x000000010016cb27 python`call_function + 183
    frame #29: 0x000000010016a88d python`_PyEval_EvalFrameDefault + 46173
    frame #30: 0x000000010015e1fe python`_PyEval_EvalCodeWithName + 414
    frame #31: 0x00000001001c1760 python`PyRun_FileExFlags + 256
    frame #32: 0x00000001001c0bd7 python`PyRun_SimpleFileExFlags + 391
    frame #33: 0x00000001001ee7bf python`pymain_main + 9583
    frame #34: 0x0000000100001bbd python`main + 125
    frame #35: 0x00007fff7de973d5 libdyld.dylib`start + 1
```

rdaly525, 2019-06-25:
This looks like a simulator issue. @dillonhuff i think cast<> is being called badly somewhere

leonardt, 2019-06-25:
Nope, the issue is in the magma coreir simulator frontend improperly translating the select path, I have a fix coming

leonardt, 2019-06-25:
Here's how I diagnosed the issue:
```
❯ git diff
diff --git a/magma/simulator/coreir_simulator.py b/magma/simulator/coreir_simulator.py
index d6abcf0..73c3093 100644
--- a/magma/simulator/coreir_simulator.py
+++ b/magma/simulator/coreir_simulator.py
@@ -53,9 +53,12 @@ def convert_to_coreir_path(bit, scope):
     # Handle renaming due to flatten types
     arrOrTuple = bit
     while isinstance(arrOrTuple.name, ArrayRef) or isinstance(arrOrTuple.name, TupleRef):
+        print(port, type(arrOrTuple))
         if isinstance(arrOrTuple, ArrayType) or isinstance(arrOrTuple, TupleType):
+            print(port)
             port, idx = port.split('.', 1)
             port += '_' + idx
+            print(port)
         if isinstance(arrOrTuple.name, ArrayRef):
             arrOrTuple = arrOrTuple.name.array
         elif isinstance(arrOrTuple.name, TupleRef):
```

When we run the test, we see that the `O._1` reference isn't being properly translated into `O__1` (like `O._0`)
```
❯ pytest tests/test_tuple.py -s -k test_atomTupleAppender
/Users/lenny/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
===================================================================== test session starts ======================================================================
platform darwin -- Python 3.7.2, pytest-4.2.0, py-1.8.0, pluggy-0.9.0
rootdir: /Users/lenny/repos/aetherling, inifile:
plugins: parallel-0.0.9, cov-2.6.1, codestyle-1.4.0
collecting ... /Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Starting topological sort
topo_order.size() = 4
numVertices(g)    = 4
O._0 Array[8, In(Bit)]
O._0
O__0
O._1 In(Bit)
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file /Users/lenny/repos/coreir/src/../include/coreir/ir/casting/casting.h, line 249.
Fatal Python error: Aborted
```

David-Durst, 2019-06-25:
I believe https://github.com/phanrahan/magma/pull/418 has unfixed this issue. See https://travis-ci.com/David-Durst/aetherling/jobs/210958278#L3049-L3055

leonardt, 2019-06-25:
yes, I'm able to reproduce it locally, investigating, it looks like it's failing trying to select a different port now

leonardt, 2019-06-25:
Hotfix in https://github.com/phanrahan/magma/pull/420, can you try it out?

David-Durst, 2019-06-25:
I'm running it on my local machine. Looks good so far but will approve the PR when all the tests pass locally.

closed 2019-07-10 by David-Durst