phanrahan/magma #419: Magma 0.9

phanrahan, 2019-06-25:
Tentative features for next major release of Magma

- [ ] New type system
   - [ ] bracket syntax Bits[n] instead of Bits(n)
   - [ ] Sum and product types
   - [ ] Consistent with hwtypes
- [ ] Better coreir integration
- [ ] Consistent mantle implementations and tests
- [ ] Uniquification
- [ ] New generator framework
- [ ] Updated magmathon

leonardt, 2019-06-25:
I added two things to start making things more concrete.

For "better coreir integration", one thing that I have been thinking about: 
>  Add a mode to compile circuit combinational/sequential directly to verilog using veriloggen and pass through the black box modules to coreir

This would improve the output verilog without having to reconstruct structure (if statements) from the netlist format

For " Consistent mantle implementations and tests"
>  hwtypes implementation of mantle to test against

@cdonovick brought up that we could provide one-to-one magma/hwtypes implementations for mantle, we could use the hwtypes version for generating simulation vectors and formal (verify the mantle generated RTL is equivalent to the SMT)

phanrahan, 2019-06-26:
Good comments. Thanks.

phanrahan, 2019-06-27:
Caleb made the suggestion that we simplify the system as we move towards full support for coreir.

The suggestion was to move all the hwtypes/coreir primitives to magma. The goal is to build is to tightly integrate these primitives into magma. A secondary goal is to simplify the mantle implementation, which is complicated because the mantle primitives do mapping.

The new mantle would just contain functions built on the primitives, such as counters, shift registers, fifos, etc.