phanrahan/magma #423: Mapping Over Upsample Segfaults

David-Durst, 2019-07-10:
The following test segaults: https://github.com/David-Durst/aetherling/blob/bf36bbc1745a617b4a46852d5207a24d048ee321/tests/test_map.py#L109.

```
(aetherling) durst@DN525agm:~/dev/W17-8/aetherling$ pytest -s tests/test_map.py::test_map_merge_rv_ce
/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
============================================================================================================ test session starts =============================================================================================================
platform darwin -- Python 3.7.1, pytest-4.4.1, py-1.8.0, pluggy-0.9.0
rootdir: /Users/durst/dev/W17-8/aetherling
plugins: xdist-1.28.0, forked-1.0.2
collected 1 item

tests/test_map.py /Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Starting topological sort
topo_order.size() = 798
numVertices(g)    = 798
python(16619,0x10af685c0) malloc: can't allocate region
*** mach_vm_map(size=3458764513820540928) failed (error code=3)
python(16619,0x10af685c0) malloc: *** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
Fatal Python error: Aborted

Current thread 0x000000010af685c0 (most recent call first):
  File "/Users/durst/dev/W17-8/pycoreir/coreir/simulator.py", line 77 in execute
  File "/Users/durst/dev/W17-8/magma/magma/simulator/coreir_simulator.py", line 249 in evaluate
  File "/Users/durst/dev/W17-8/magma/magma/simulator/coreir_simulator.py", line 197 in __init__
  File "/Users/durst/dev/W17-8/aetherling/tests/test_map.py", line 119 in test_map_merge_rv_ce
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/python.py", line 178 in pytest_pyfunc_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/python.py", line 1464 in runtest
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 123 in pytest_runtest_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 198 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 226 in from_call
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 198 in call_runtest_hook
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 173 in call_and_report
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 93 in runtestprotocol
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/runner.py", line 78 in pytest_runtest_protocol
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 270 in pytest_runtestloop
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 249 in _main
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 209 in wrap_session
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/main.py", line 242 in pytest_cmdline_main
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 62 in <lambda>
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/manager.py", line 68 in _hookexec
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pluggy/hooks.py", line 289 in __call__
  File "/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/_pytest/config/__init__.py", line 79 in main
  File "/Users/durst/anaconda/envs/aetherling/bin/pytest", line 11 in <module>
Abort trap: 6
(aetherling) durst@DN525agm:~/dev/W17-8/aetherling$
```

leonardt, 2019-07-10:
From what I understand form that log, I see that it's aborting in
```
  File "/Users/durst/dev/W17-8/pycoreir/coreir/simulator.py", line 77 in execute
```

which is just a call to the coreir C api (https://github.com/leonardt/pycoreir/blob/54dc35b22320f3d105620ea899c0fe9bb3c23529/coreir/simulator.py#L77) which would suggest that the segfault is happening internally to the simulator.

`malloc: can't allocate region` seems like it is a memory related issue, do you know how much memory this test is consuming? Could it be that we're just running out of memory? Or this could point to some sort of memory leak.

leonardt, 2019-07-10:
A quick google of the `mach_vm_map` message reveals a bunch of out of memory related problems.