phanrahan/magma #444: Wrong Error Message

liyilui, 2019-10-04:
Consider following code:
`def DefineBaseModule():
    class baseModule(m.Circuit):
        name = "base_module"
        baseIoDict = {
        "in0":m.In(m.Bit),
        "out0":m.Out(m.Bit)
        }
        IO = ["baseIO",m.Tuple(**baseIoDict)]
    return baseModule

def DefineHier():
    class HierModule(m.Circuit):
        name = "hier_module"
        #IO = ["hier",hierIO()]
        #IO = ["baseIO",baseIO()]
        baseIoDict = {
        "in0":m.In(m.Bit),
        "out0":m.Out(m.Bit)
        }
        IO = ["baseIO", m.Tuple(**baseIoDict)]
        @classmethod
        def definition(io):
            baseM = DefineBaseModule()
            m.wire(getattr(baseM,"baseIO"),getattr(io,"baseIO"))
    return HierModule

baseMH = DefineHier()`

The main error here is DefineBaseModule() [we should use DefineBaseModule()()], where we try to wire to a class instead of to an instance. But the error message is saying 
`
magma:ERROR:Using `hier_module.baseIO.in0` (an output) as an input
magma:ERROR:Using `hier_module.baseIO.out0` (an input) as an output
`
which is confusing.
Could you help add something there to make error message easier to debug? Thanks a lot!

leonardt, 2019-10-07:
Yes, we should add a better error message here. The problem is that the user is trying to wire up a definition (generated by a function) rather than an instance of the generated definition (common issue when forgetting the parens).  I think we should be able to notice that we're currently inside a definition (HierModule in this case), and raise an error if trying to wire up a port for any other definitions

leonardt, 2019-10-07:
Should be fixed by https://github.com/rdaly525/coreir/releases/tag/v0.1.20
See https://github.com/phanrahan/magma/pull/450 for the test case, can you confirm and close the issue? Thanks!

leonardt, 2019-10-07:
Edit, ignore my last comment, that was for another issue.