phanrahan/magma #445: Cannot support verilog dump for nested tuple-tuple io connection

liyilui, 2019-10-04:
def hierIO():
    dictIO = {
        "baseIO":baseIO(),
        "ctr": m.In(m.Bit)
    }
    return m.Tuple(**dictIO)

def DefineCtrModule():
    class ctrModule(m.Circuit):
        name = "ctr_module"
        IO = ["ctr",m.In(m.Bit)]
        # @classmethod
        # def definition(io):
        #     mantle.coreir.DefineTerm(1)().I <= io.ctr
    return ctrModule

def baseIO():
    dictIO = {
        "in0":m.In(m.Bit),
        "out0":m.Out(m.Bit)
    }
    return m.Tuple(**dictIO)

def DefineBaseModule():
    class baseModule(m.Circuit):
        name = "base_module"
        # baseIoDict = {
        # "in0":m.In(m.Bit),
        # "out0":m.Out(m.Bit)
        # }
        IO = ["baseIO",baseIO()]
        #@classmethod
        # def definition(io):
        #     mantle.coreir.DefineTerm(1)().I[0] <= io.baseIO.in0
        #     #m.wire(io.baseIO.out0,mantle.coreir.DefineTerm(1)().I[0])
    return baseModule

def DefineHier():
    class HierModule(m.Circuit):
        name = "hier_module"
        # #IO = ["hier",hierIO()]
        # #IO = ["baseIO",baseIO()]
        # baseIoDict = {
        # "in0":m.In(m.Bit),
        # "out0":m.Out(m.Bit)
        # }
        IO = ["hier", hierIO()]
        @classmethod
        def definition(io):
            baseM = DefineBaseModule()()
            ctrM = DefineCtrModule()()
            m.wire(baseM.baseIO,io.hier.baseIO)
            #m.wire(getattr(baseM,"baseIO"),getattr(io,"baseIO"))
            m.wire(ctrM.ctr,io.hier.ctr)
    return HierModule

baseMH = DefineHier()
m.compile("baseMH", baseMH, output="coreir-verilog")

When I run this, I got error:
/Users/travis/build/rdaly525/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/travis/build/rdaly525/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/travis/build/rdaly525/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
ERROR: NYI flatten types of generator or nodef module
{global.base_module}.baseIO Is not a flattened type!
  Type is: {'in0':BitIn, 'out0':Bit}

0   libcoreir.dylib                     0x00000001076881c6 _ZN6CoreIR6Passes12FlattenTypes22runOnInstanceGraphNodeERNS_17InstanceGraphNodeE + 278
1   libcoreir.dylib                     0x0000000107380213 _ZN6CoreIR11PassManager20runInstanceGraphPassEPNS_4PassE + 451
2   libcoreir.dylib                     0x00000001073808ae _ZN6CoreIR11PassManager7runPassEPNS_4PassERNSt3__16vectorINS3_12basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEENS8_ISA_EEEE + 1086
3   libcoreir.dylib                     0x0000000107382191 _ZN6CoreIR11PassManager3runERNSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 1969
4   libcoreir.dylib                     0x00000001071d49c8 _ZN6CoreIR7Context9runPassesENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 136
5   coreir                              0x000000010703e194 main + 15300
6   libdyld.dylib                       0x00007fff7f0a73d5 start + 1

leonardt, 2019-10-07:
This looks like a bug in the internals, investigating.

leonardt, 2019-10-07:
@liyilui the issue is that coreir has not yet implemented support for compiling a tuple type on a declaration (no definition). I will add support for this, but for now, a workaround to unblock you is to wire up the ports of the circuit so it has a definition (not a declaration). The culprit is DefineBaseModule, here is a temporary fix:
```
    def DefineBaseModule():
        class baseModule(m.Circuit):
            name = "base_module"
            IO = ["baseIO",baseIO()]
            @classmethod
            def definition(io):
                m.wire(io.baseIO.out0, io.baseIO.in0)
        return baseModule
```

I will let you know when support has been added for compiling tuples in declarations.

leonardt, 2019-10-07:
GH issue on coreir tracking this: https://github.com/rdaly525/coreir/issues/800

leonardt, 2019-10-07:
Should be fixed by https://github.com/rdaly525/coreir/releases/tag/v0.1.20 can you try updating your coreir and seeing if it works?
See https://github.com/phanrahan/magma/pull/450 for a test case

closed 2019-10-07 by leonardt