phanrahan/magma #448: Array length

phanrahan, 2019-10-07:
Most of the loam tests fail.

For example,
```
% cd loam
% py.test --target ice40    tests/test_mantle/arith/test_arith.py 
```
Fails on 
```
    def __init__(self, *largs, **kwargs):
    
        Type.__init__(self, **kwargs)
    
>       if isinstance(largs, Sequence) and len(largs) == self.N:
E       AttributeError: 'Bits' object has no attribute 'N'

../magma/magma/array.py:91: AttributeError
```
Seems like Bits is not a subclass of Array any longer.

leonardt, 2019-10-07:
I think I've seen this error when using the old syntax with parens (e.g. `Bits(N)`), can you double check if we've updated the code to the `Bits[n]` style bracket syntax?

phanrahan, 2019-10-07:
Found an old-style Bits(n) in loam/fpga.py. Seems to work better now.

phanrahan, 2019-10-07:
Should we depreciate `Bits(n)`?

leonardt, 2019-10-07:
@phanrahan yes, that interface is no longer supported by `hwtypes` so magma should not be using it either