phanrahan/magma #449: python simulator and magmathon

phanrahan, 2019-10-07:
I am trying to run the following program from the magmathon.
```
from magma.simulator import PythonSimulator

fulladder_magma = PythonSimulator(FullAdder)

assert fulladder_magma(1, 0, 0) == fulladder(1, 0, 0), "Failed"
print("Success!")
```
This fails:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-5-16bac3e07308> in <module>
      3 fulladder_magma = PythonSimulator(FullAdder)
      4 
----> 5 assert fulladder_magma(1, 0, 0) == fulladder(1, 0, 0), "Failed"

~/git/magma/magma/simulator/python_simulator.py in __call__(self, *largs)
    390                 j += 1
    391 
--> 392         self.evaluate()
    393 
    394         outs = []

~/git/magma/magma/simulator/python_simulator.py in evaluate(self, no_update)
    333             primitive.simulate()
    334         for primitive in self.execution_order.combinational:
--> 335             primitive.simulate()
    336 
    337         triggered = []

~/git/magma/magma/simulator/python_simulator.py in simulate(self)
     65 
     66     def simulate(self):
---> 67         self.primitive.simulate(self.value_store, self.state_store)
     68 
     69 class WatchPoint:

~/git/mantle/mantle/coreir/logic.py in simulate(self, value_store, state_store)
     39         I0 = ht.Bit(value_store.get_value(self.I0))
     40         I1 = ht.Bit(value_store.get_value(self.I1))
---> 41         O = python_op(I0, I1).as_bool_list()[0]
     42         value_store.set_value(self.O, O)
     43 

AttributeError: 'Bit' object has no attribute 'as_bool_list'
```

leonardt, 2019-10-07:
The python simulator uses a lists of bools to represent bitvectors internally.  The longterm fix should be to update the code to just use BitVector internally, but a hotfix would be to either create a helper function to reimplement `as_bool_list` (this interface was dropped when we updated the BV class), or add the method back to the BV API.  @cdonovick do you happen to know if there's a method to convert a BitVector to a list of `bool`s?

cdonovick, 2019-10-08:
There is not.  If we are happy with `as_bit_list` and not `as_bool_list` we could add it to the api its just `[self[i] for i in range(self.size)]`.  However, `as_bool_list` cannot be supported by SMT or magma and hence will not be a method of `AbstractBitVector`.  In general I am resistant to adding methods to `BitVector` that are not a part of `AbstractBitVector` as people will use them assuming they work for all BV types.

Now usually I am a consenting adults kind of guy and don't feel the need to protect people from such errors, but our users have historically failed at being adult (see: `__bool__` and `__int__`).  Then again we already have `as_uint` / `as_sint` so maybe we can get away with add `as_bool_list` without people trying to use it in peak / magma code.

cdonovick, 2019-10-08:
Wait I lied. `BitVector` still has as_bool_list.

cdonovick, 2019-10-08:
The issue is `Bit` does not have `as_bool_list` which to me seems like a silly method to put on a `Bit` type.

leonardt, 2019-10-08:
Using hwtypes internally in the simulator is likely the best long term solution. I agree managing the conversions between these various representations makes maintenance very difficult

phanrahan, 2019-10-08:
I removed the `as_bool_list`. Tests work. Still haven't converted everything to use hwtypes.

closed 2019-10-08 by phanrahan