phanrahan/magma #452: Anonymous type wiring order

rsetaluri, 2019-10-09:
```
import magma as m

class Foo(m.Circuit):
    IO = ["I", m.In(m.Bit), "O", m.Out(m.Bit)]

    @classmethod
    def definition(io):
        anon = m.Bit()
        #m.wire(anon, io.I)                                                                                                                            
        m.wire(io.I, anon)
        m.wire(anon, io.O)
```

In the above code, if we swap lines 9 and 10 (i.e. swap the order of wiring of an anonymous bit), we get the error: `magma:ERROR:Using `Foo.I` (an output) as an input`.

Is this expected/desired behavior? For all other wiring calls the system figures out the proper semantics (making the order of wiring irrelevant).

leonardt, 2019-10-09:
I think the issue is that an anonymous wire acts both as an input and an output, triggering this error message. One thing to check is whether the circuit is compiling correctly still, if so, this may be a false positive error message that we can adjust in the case that one of the bits is anonymous. If it's not, we will need to improve the logic to handle this case.

phanrahan, 2019-10-10:
Hmm. I just tried this and both orders work for me.

rsetaluri, 2019-11-01:
@phanrahan - I'm still seeing this issue. Is this the exact code you're trying, on master?

```python
import magma as m

class Foo(m.Circuit):
    IO = ["I", m.In(m.Bit), "O", m.Out(m.Bit)]

    @classmethod
    def definition(io):
        anon = m.Bit()
        m.wire(anon, io.I)
        m.wire(anon, io.O)

m.compile("Foo", Foo, output="coreir")
```

phanrahan, 2019-11-01:
I see. I tried wiring in different orders. This works:
```
 m.wire(io.I, anon)
 m.wire(anon, io.O)
```
But this does not.
```
m.wire(anon, io.I)
m.wire(anon, io.O)
```
I will look into it.