phanrahan/magma #453: DefineFromVerilog Fails

David-Durst, 2019-10-10:
https://github.com/David-Durst/aetherling/blob/b2205ae7a40e2118e1af23e1a359a4c01ad9857a/aetherling/examples/ae_circuit24998-300.py fails on line 12 when loading the verilog file https://github.com/David-Durst/embeddedHaskellAetherling/blob/rewrites/test/verilog_examples/aetherling_copies/map_s_4.v that was generated by coreir. The message is
```
Syntax error
Traceback (most recent call last):
  File "/home/david/dev/aetherling/aetherling/examples/ae_circuit24998-300.py", line 12, in <module>
    mod = Main()
  File "/home/david/dev/aetherling/aetherling/examples/ae_circuit24998-300.py", line 6, in <lambda>
    Main = lambda : m.DefineFromVerilog("/home/david/dev/embeddedHaskellAetherling/test/verilog_examples/aetherling_copies/map_s_4.v", target_modules=["top"])[0]
  File "/home/david/dev/magma/magma/fromverilog.py", line 251, in DefineFromVerilog
    shallow=shallow, external_modules=external_modules)
  File "/home/david/dev/magma/magma/fromverilog.py", line 169, in FromVerilog
    ast = parser.parse(source)
  File "/home/david/anaconda3/envs/ae2/lib/python3.7/site-packages/pyverilog/vparser/parser.py", line 71, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
  File "/home/david/anaconda3/envs/ae2/lib/python3.7/site-packages/pyverilog/vparser/ply/yacc.py", line 265, in parse
    return self.parseopt_notrack(input,lexer,debug,tracking,tokenfunc)
  File "/home/david/anaconda3/envs/ae2/lib/python3.7/site-packages/pyverilog/vparser/ply/yacc.py", line 1047, in parseopt_notrack
    tok = self.errorfunc(errtoken)
  File "/home/david/anaconda3/envs/ae2/lib/python3.7/site-packages/pyverilog/vparser/parser.py", line 2214, in p_error
    self._coord(p.lineno))
  File "/home/david/anaconda3/envs/ae2/lib/python3.7/site-packages/pyverilog/vparser/plyparser.py", line 55, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pyverilog.vparser.plyparser.ParseError: :1: before: /
```

@leonardt thoughts?

THofstee, 2019-10-10:
`pip install -e git+git://github.com/pyhdi/pyverilog.git#egg=pyverilog`.

The pyverilog example parser parses this code with no issues for me.

David-Durst, 2019-10-10:
Can I use that one instead? Is magma using a custom parser?

THofstee, 2019-10-10:
The PyPI version of pyverilog (which magma uses) isn't kept up to date with master that well. Maybe there was a commit fixing behavior between the latest release and now. Or you have a very old version installed.

David-Durst, 2019-10-10:
Turns out that wasn't the issue. I was calling `DefineFromVerilog` with a path to a file. I should've called `DefineFromVerilogFile`.

A new issue has appeared. @rdaly525 thoughts?
```
ERROR: Module: global.top
  Type: {'CLK':BitIn, 'O':Bit[8], 'hi':BitIn[8], 'valid_down':Bit, 'valid_up':BitIn}
  Def? No has no def!

0   libcoreir.dylib                     0x00000001054bfa08 _ZN6CoreIR7Context6setTopEPNS_6ModuleE + 104
1   libcoreir.dylib                     0x0000000105579bf6 _ZN6CoreIR12loadFromFileEPNS_7ContextENSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEPPNS_6ModuleE + 29638
2   coreir                              0x0000000105327cc3 main + 9587
3   libdyld.dylib                       0x00007fff610093d5 start + 1
Traceback (most recent call last):
  File "/Users/durst/dev/W17-8/fault/fault/verilog_target.py", line 49, in __init__
    raise Exception(f"Compiling {self.circuit} failed")
Exception: Compiling top(CLK: In(Bit), O: Out(Bits[8]), hi: In(Bits[8]), valid_down: Out(Bit), valid_up: In(Bit)) failed
```

leonardt, 2019-10-10:
Are you trying to compile a top circuit that is defined from verilog? IIRC there may be a bug related to that in coreir, for a workaround, could you try setting `magma_output="verilog"` as a parameter to the fault compile_and_run command? This should tell magma to use the verilog backend which hopefully should work since it's just dumping out the verilog file.

David-Durst, 2019-10-10:
That works! Thank you

closed 2019-10-10 by David-Durst