phanrahan/magma #454: [feature request] Pattern for simulation only logic

leonardt, 2019-10-10:
One feature request is to support the ability to define logic in a module that is only used for simulation purposes. The simplest way on the user's part would be to write the code in the definition, and mark the instances/ports as "simulation only". We could then have a magma pass that removes these instances/ports when compiling with a "synthesis" option. Thoughts on other designs for this?

leonardt, 2019-10-10:
A way to do this now without any built-in language support would be to use a generator parameter to guard the inclusion of code (basically a Python level "ifdef")
```python
def gen(include_debug=False):
    class Circ(m.Circuit):
        ...
        def definition(io):
            ...
            if include_debug:
                <insert debug logic>
```

rsetaluri, 2019-10-15:
I think `debug` as a generator flag is actually not a bad pattern. However, we could try to enable this is a first class citizen. The idea is to leverage the fact that we can define more functions on the class than just `definition`. For example:

```
class Circ(m.Circuit):
     @classmethod
    def definition(io):
        # some normal circuit logic

    def debug(io):
        # some extra debug circuit logic
```

Then at definition time we could choose to call `cls.debug()` or not based on some parameter. Issues with this:

1. The switch is at definition time, not compile time. It would have to be something like `m.enable_debugging()` at the top of your main. However, this may not be so bad since if you want to wire a debug port, it has to be present at definition time.
2. All debug features have to be additive (not subtractive or mutative).
3. Adding ports in the `debug()` function is not immediately available. However, if we add the ability to extend the interface this would be possible (something we want to do anyway).

Just an idea.