phanrahan/magma #457: Wiring allows non-matching types to be wired

jameshegarty, 2019-10-14:
Ports with bit type, uint type, and sint type can be wired to each other, as long as the length matches. Don't we want to check that they type itself matches, and require an explicit cast?

phanrahan, 2019-10-15:
Currently,

Wiring is structural. So, if two types have the same structure, then they can be connected.

If you want to override the behavior or wiring, you can overload the operator on the class to be stricter.

Operators depend on the type. For example, they can add checks the types.

Open to suggestions.

jameshegarty, 2019-10-15:
It seem like right now, if you have a signed add module, you can send unsigned inputs to it and it will wire it without an error. IMO I would prefer it to give an error in this case unless you do an explicit cast. A lot of problems of Verilog come up because it doesn't do typechecking like this. Thoughts anybody?

THofstee, 2019-10-15:
Except wiring is _not_ structural, because I can't wire `Bit` to `Bits[1]`.

phanrahan, 2019-10-15:
Wiring a scalar to an array is not normally considered structural.

phanrahan, 2019-10-15:
I agree that the add should do type checking. Let me look at the code.

jameshegarty, 2019-10-15:
That'd be cool, I'd be happy to help too once I clear my plate of some other stuff :)

phanrahan, 2019-10-15:
Let's move this discussion to hwtypes. And get @leonardt and @cdonovick involved.

cdonovick, 2019-10-15:
I think this discussion would fit well under https://github.com/leonardt/hwtypes/issues/98
Although we can  also open a new issue.