phanrahan/magma #459: All types are equal?

jameshegarty, 2019-10-15:
All of the types appear to be equal to each other? is there some way to check which type a given type (or value) is?

ie:
print("eq?",m.UInt[2]==m.SInt[2])
prints true

THofstee, 2019-10-15:
You might be able to do `type(m.UInt[2]) == type(m.SInt[2])`

leonardt, 2019-10-15:
Yea, trying to do equality demonstrates some interesting (unexpected) behavior. This should be resolved by the pending changes in the `hwtypes` branch which updates a lot of this related machinery.

For now, you can use `is` since types are cached, so the objects should be the same, e.g.

```python
>>> m.UInt[2] is m.UInt[2]
True
>>> m.UInt[2] is m.SInt[2]
False
```

leonardt, 2019-10-15:
Related to @THofstee 's suggestion, another common pattern is to check the kind of a type, e.g.
```
>>> isinstance(m.UInt[2], m.UIntKind)
True
>>> isinstance(m.UInt[2], m.SIntKind)
False
```

phanrahan, 2019-10-15:
It should be possible to support the == operator. These changes are hopefully pending.

In the meantime, we have workarounds.

closed 2020-09-22 by leonardt