phanrahan/magma #465: Signed multiply primitive and multiply primitive input vs output width

leonardt, 2019-10-24:
It's come to our attention that the inclusion of a signed multiply primitive may be beneficial for generating efficient hardware.  The current magma/coreir multiply primitive does not distinguish between signed/unsigned, and requires the user to extend the inputs to the desired output width (z or s ext depending on the input type).  This is semantically correct, but there are cases where one can implement a more efficient circuit when doing the "sign extend" logic as part of the multiplication circuit.  The same goes for a multiply circuit with different input widths (where the output width is input0_width + input1_width).

There seems to be two options here:
* preserve the current semantics, and implement a compiler optimization passes that strips these bit width extension operations and replaces them with a specialized circuit that implements the extension as part of the operation
* adjust the multiply definition to distinguish between signed/unsigned (and also allow for variable input widths) so that the specialized primitive is generated by default.

One key point is that the signed multiply primitive can be generically defined in terms of the base coreir primitives (extension + multiply), so this need not require a change to the coreir semantics, but may be considered (it also has the issue of being inconsistent with the SMT-LIB bitvector primitives, but again can be defined in terms of the base operators).

This brings up another question about the operators in general, e.g. including the extra carry bit in the output of add (versus extending the inputs to get a carry).  It seems like there are two key issues: ergonomics (does the user have to do extensions whenever they want a carry) and optimizations (does extending the input values to get the desired result have any affect on the final quality of the circuit compared to a specialized implementation that does the extension internally?).  We can always define multiple implementations of the operations (e.g. Add and AddC) which have different semantics w.r.t. to the output width, but it's worth consider what the default semantics for the built-in operators are (e.g. `+`).  

Right now it seems that having the operators not do any implicit changes to the width (i.e. the width of the outputs is equal to the width of the inputs) and having special (explicit) functions for the variants that do may be a good option.  This has the least amount of implicit behavior, but requires a little more work on the user side (need to say "AddC" instead of "+" when wanting the carry bit).

Thoughts?

In theory I think it should be possible to do an optimization pass that removes the unnecessary extensions and replaces them with a specialized circuit, so that may be a viable longer term solution, however the way that magma handles extensions (anonymously) could be changed to facilitate this (insert explicit extension instances so they are easier to find), but perhaps there is a sophisticated pattern we can use to detect these.  However, this also begs the questions as to what the best user interface is (do we want them always extending the inputs for multiplies?)

leonardt, 2019-10-24:
CC @rdaly525

phanrahan, 2019-10-25:
I understand their view. In terms of implementation, these are quite different.

I suggest we talk to @cdonovick and Clark about the best way to do this.

My opinion is that it is ok to include non-smt-bitvector primitives. But when we do, we should be careful to include a definition in terms of standard smt primitives.

cdonovick, 2019-10-25:
I have no problems with adding a methods to `AbstractBitVector` as long a mxin implementation can be provided.  
We currently have `BV.adc :: BV[n] -> BV[n] -> Bit -> (BV[n], BIT)` .  I see no problem with adding `AddExt :: BV[n] -> BV[n] -> BV[n+1]` and  `MulExt :: BV[n] -> BV[n] -> BV[2*n]`.  Although I think the operators should continued to be defined as is.  If a user wants to change the operators they can always do something like `BV.__add__ = BV.AddExt`

phanrahan, 2019-10-29:
I agree with @cdonovick

rdaly525, 2019-10-29:
I agree with defining these as additional methods instead of changing the default semantics of add/mul.  I would also advocating defining these circuits in mantle.