phanrahan/magma #471: WARNING:root:Register.reg_PR_inst0.arst not connected

hofstee, 2019-10-29:
```
import magma as m

def DefineRegister__(width : int):
    @m.circuit.sequential(async_reset=True)
    class Register:
        def __init__(self):
            self.value: m.Bits[width] = m.bits(0, width)

        def __call__(self, I: m.Bits[width]) -> m.Bits[width]:
            O = self.value
            self.value = I
            return O

    return Register

m.compile("asdf", DefineRegister__(16))
```

Shouldn't this work? It's basically the listed example. This is the output verilog:

```
module Register_comb (input [15:0] I, input [15:0] self_value_O, output [15:0] O0, output [15:0] O1);
assign O0 = I;
assign O1 = self_value_O;
endmodule

module Register (input [15:0] I, input  CLK, input  ASYNCRESET, output [15:0] O);
wire [15:0] reg_PR_inst0_out;
wire [15:0] Register_comb_inst0_O0;
wire [15:0] Register_comb_inst0_O1;
coreir_reg_arst #(.init(0)) reg_PR_inst0 (.I(Register_comb_inst0_O0), .clk(CLK), .O(reg_PR_inst0_out));
Register_comb Register_comb_inst0 (.I(I), .self_value_O(reg_PR_inst0_out), .O0(Register_comb_inst0_O0), .O1(Register_comb_inst0_O1));
assign O = Register_comb_inst0_O1;
endmodule
```

leonardt, 2019-10-29:
Note added via c9dc57c

Let me know if the "verilog" target is required for your flow, otherwise I'll keep this open to track the issue but it will be low priority.

leonardt, 2020-08-07:
Same as https://github.com/phanrahan/magma/issues/472

closed 2020-08-07 by leonardt