phanrahan/magma #472: AttributeError: 'AnonRef' object has no attribute 'tuple'

hofstee, 2019-10-29:
```
@m.circuit.sequential(async_reset=True)
class Counter:
    def __init__(self):
        self.count : m.UInt[16] = m.uint(0, 16)

    def __call__(self, inc : m.Bit) -> m.UInt[16]:
        if inc:
            self.count = self.count + 1
        else:
            self.count = self.count

        O = self.count
        return O
```

What's going on here?

leonardt, 2019-10-29:
Looks like an issue again with the verilog backend, see the followup to https://github.com/phanrahan/magma/issues/471

```
import magma as m

@m.circuit.sequential(async_reset=True)
class Counter:
    def __init__(self):
        self.count : m.UInt[16] = m.uint(0, 16)

    def __call__(self, inc : m.Bit) -> m.UInt[16]:
        if inc:
            self.count = self.count + 1
        else:
            self.count = self.count

        O = self.count
        return O

m.compile("Counter", Counter, output="coreir-verilog")
```

seems to compile fine for me.

I think I'll propose changing the default compile target to "coreir-verilog" which should help avoid this issue in the future, sorry about that.

leonardt, 2020-08-07:
Default option is coreir-verilog, we have no plans to support this with the verilog backend.

closed 2020-08-07 by leonardt