phanrahan/magma #473: False positive error in unconnected ports for clock types

leonardt, 2019-10-29:
For this example:
```
import magma as m

@m.circuit.sequential(async_reset=True)
class Counter:
    def __init__(self):
        self.count : m.UInt[16] = m.uint(0, 16)

    def __call__(self, inc : m.Bit) -> m.UInt[16]:
        if inc:
            self.count = self.count + 1
        else:
            self.count = self.count

        O = self.count
        return O

m.compile("Counter", Counter, output="coreir-verilog")
```

magma issues
```
magma:ERROR:Input port reg_PR_inst0.clk not driven
magma:ERROR:Input port reg_PR_inst0.arst not driven
```

however this compiles okay because the coreir backend will automatically wire up clock/reset ports.  Perhaps in the unconnected check we can issue a warning or log about clock/reset types (instead of error) and note that the backend will automatically wire them up? Or we can run the wiring pass at the end of definition time instead of in the backend (this could be better to enforce consistency among the backends). Then we wouldn't need to change the error check logic.

phanrahan, 2019-10-29:
I don't think this should be an error. I also don't see anything wrong with using the default clock wiring logic. To me that is a feature.

phanrahan, 2019-10-31:
I would like to remove this check. It is causing lots of warnings in my code.

hofstee, 2019-11-08:
Also causing a lot of noise on GarnetFlow: https://buildkite.com/stanford-aha/garnetflow/builds/880#305a3c5c-70f4-46c8-ac4d-b3663a6e4de2/407-4694

rsetaluri, 2019-11-08:
@hofstee @phanrahan will fix this by moving this check to *after* the automatic clock/reset wiring pass. will have something today.

rsetaluri, 2019-11-08:
Fixed by #489

closed 2019-11-08 by rsetaluri