phanrahan/magma #476: m.uint(Bit, n>1) is unintuitive

hofstee, 2019-10-29:
If you have a m.Bit, and you call `m.uint(bit, 16)` for example, it replicates the bit to all 16 bits, effectively doing sign-extension. For a uint I think the behavior I would expect is that the bit passed in is zero-extended, not sign-extended. For m.sint I would probably expect the sign-extension behavior.

leonardt, 2019-10-31:
I believe this is an artifact of the case in the `array` constructor that allows you to initialize an array using a single value, e.g. `Array[5](0)` creates an array of zeros, similar to how you could say `int [5] arr = {0}` in C.  I'll have to review exactly whats going on inside `m.uint`, but I suspect that it's dispatching to that logic.  I think we can preserve the current constructor interface, but for the conversion functions I do think that doing a zext or sext of the value makes more sense, so perhaps we can just add this behavior in the conversion logic.

phanrahan, 2019-10-31:
I agree that we should do a sext or zext in this case.

rsetaluri, 2019-11-01:
I also agree we should have to do a sext or zext. But I think then we should disallow `m.uint(bit, 16)` as a conversion - it is confusing. If we want this capability we should have a magma function like `fill`.

phanrahan, 2019-11-01:
Good point. If I understand you, we would only allow:
```
m.uint(int,  16)
m.uint(Bits[16])
```
And provide a `fill` for a `Bit`.

rsetaluri, 2019-11-01:
Yes, I would suggest that proposal.

closed 2020-01-16 by leonardt