phanrahan/magma #478: An argument for zext, sext

hofstee, 2019-10-30:
Currently `zext` zero-extends by adding N bits to the already present number of bits. The [zfill](https://docs.python.org/3.8/library/stdtypes.html?highlight=zfill#str.zfill) function in python makes the result N bits wide (and returns the original string if N is less than its existing width). That's one argument for making `zext` behave in a similar manner. I assume `sext` follows suit.

phanrahan, 2019-10-30:
Originally zext worked as you propose. Several people recommended the current way of working. I think we should keep it the way it is so that the api does not thrash.

rsetaluri, 2019-11-01:
Agree w/ @phanrahan - if it is convenient to do it that way @hofstee, I would suggest a user function which takes in that argument and does the arithmetic to dispatch to magma.zext/sext.

closed 2019-11-01 by rsetaluri