phanrahan/magma #479: N-d array doesn't work

Kuree, 2019-10-31:
Using the master branch of leonardt/pyverilog and magma. Can't create find the issues tab there so creating one here.

Below is the minimal example to reproduce it:
```
import magma as m

src = """module aggregator (
  agg_out,
  clk,
  in_pixels,
  rst,
  valid
);
output logic [7:0] agg_out [7:0];
input logic  clk;
input logic [7:0] in_pixels;
input logic  rst;
output logic  valid;
endmodule   // aggregator
"""
print(src)

m.DefineFromVerilog(src, type_map={"clk": m.In(m.Clock), "rst": m.In(m.AsyncReset), "in_pixels": m.In(m.Bits[8]), "agg_out": m.Out(m.Array[8, m.Bits[8]]), "valid": m.Out(m.Bit)})
```

Error:
```
Traceback (most recent call last):
  File "test.py", line 19, in <module>
    m.DefineFromVerilog(src, type_map={"clk": m.In(m.Clock), "rst": m.In(m.AsyncReset), "in_pixels": m.In(m.Bits[8]), "agg_out": m.Out(m.Array[8, m.Bits[8]]), "valid": m.Out(m.Bit)})
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 270, in DefineFromVerilog
    param_map=param_map)
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 201, in FromVerilog
    parsed_name, args = ParseVerilogModule(verilog_defn, type_map)
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 169, in ParseVerilogModule
    args.append(get_type(sub_child, type_map, param_map))
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 135, in get_type
    typ = convert(typ, type_map[io.name])
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 87, in convert
    raise NotImplementedError(f"Conversion between {input_type} and "
NotImplementedError: Conversion between Array[8, Out(Bits[8])] and Array[8, Out(Bits[8])] not supported
```

It seems like the parsing result is incorrect.

Using newer port declaration style yields a different error:
```
import magma as m

src = """module aggregator (
output logic [7:0] agg_out [7:0],
input logic  clk,
input logic [7:0] in_pixels,
input logic  rst,
output logic  valid);
endmodule   // aggregator
"""
print(src)

m.DefineFromVerilog(src, type_map={"clk": m.In(m.Clock), "rst": m.In(m.AsyncReset), "in_pixels": m.In(m.Bits[8]), "agg_out": m.Out(m.Array[8, m.Bits[8]]), "valid": m.Out(m.Bit)})
```
Error:
```
Traceback (most recent call last):
  File "test.py", line 13, in <module>
    m.DefineFromVerilog(src, type_map={"clk": m.In(m.Clock), "rst": m.In(m.AsyncReset), "in_pixels": m.In(m.Bits[8]), "agg_out": m.Out(m.Array[8, m.Bits[8]]), "valid": m.Out(m.Bit)})
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 270, in DefineFromVerilog
    param_map=param_map)
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 201, in FromVerilog
    parsed_name, args = ParseVerilogModule(verilog_defn, type_map)
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 153, in ParseVerilogModule
    args.append(get_type(io, type_map, param_map))
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 135, in get_type
    typ = convert(typ, type_map[io.name])
  File "/home/keyi/workspace/magma/magma/fromverilog.py", line 87, in convert
    raise NotImplementedError(f"Conversion between {input_type} and "
NotImplementedError: Conversion between Array[8, Out(Bits[8])] and Array[8, Out(Bits[8])] not supported
```
The type looks the same to me?

In both case removing the type map for regular ports works.

leonardt, 2019-10-31:
Looks like a bug, we'll investigate, for now I was able to workaround the issue by removing the array ports from the type map, so
```python 
import magma as m

src = """module aggregator (
output logic [7:0] agg_out [7:0],
input logic  clk,
input logic [7:0] in_pixels,
input logic  rst,
output logic  valid);
endmodule   // aggregator
"""
print(src)

circ = m.DefineFromVerilog(src, type_map={"clk": m.In(m.Clock), "rst": m.In(m.AsyncReset)})
print(circ)
```

closed 2019-10-31 by rsetaluri

Kuree, 2019-10-31:
What about the first bug where an unpacked array is inferred as a scalar?
Or it's not an issue at all?