phanrahan/magma #481: tests/test_constrained_random.py fails

phanrahan, 2019-10-31:
This test fails with the following error.
```
    def test_constrained_random():
        random.seed(0)
        v = dict(x=WIDTH, y=WIDTH, z=WIDTH)
    
        def pred(x, y, z):
            return (((x + y) ^ z) >> 1) == WIDTH
    
        gen = ConstrainedRandomGenerator()
    
        models = gen(v, pred, N)
>       assert len(models) >= N
E       assert 0 >= 8
E        +  where 0 = len(set())

tests/test_constrained_random.py:18: AssertionError
----------------------------- Captured stderr call -----------------------------
exception in tactic canceled
```

leonardt, 2019-11-01:
I believe this is a fault issue? Is there any more output from the test failure?

@cdonovick do you happen to know anything about that "exception in tactic canceled" error message? This is related to the SMT constrained random input generator

cdonovick, 2019-11-01:
That certainly sounds like a z3 exception but I have never seen it.  @phanrahan what version of z3 do you have installed?

phanrahan, 2019-11-01:
I just updated z3 using pip. That seems to solve the problem.
```
z3                                 0.2.0     
z3-solver                     4.8.5.0   
```

Now, the only failing test is because of `msat`. Any suggestions on how to install `msat` on my mac? `pysmt-install --msat` fails.

cdonovick, 2019-11-01:
Just fyi `z3` is a stale package which I assume the z3 people lost the keys to.  `z3-solver` is what you need installed.

cdonovick, 2019-11-01:
No idea about installing mathsat on mac. @makaimann might know.

phanrahan, 2019-11-01:
@leonardt how do we install `msat` using travis? I can't figure out how that is done.

makaimann, 2019-11-01:
That's a bummer that `pysmt-install --msat` fails on mac. Maybe we should make an issue on their repository? In the meantime, you can just `wget` it from the downloads page [here](http://mathsat.fbk.eu/download.html). MathSAT is not open-source so they just ship a library and binary.

phanrahan, 2019-11-02:
@makaimann, I installed from the downloads page. But I am still having problems. Any chance you could help me debug the issues next week?

makaimann, 2019-11-02:
Sorry that's been such a pain. I'm happy to help debug! I'll be around every day next week.

leonardt, 2019-11-02:
It looks like we only install msat on travis for linux so this has been a persistent issue: https://github.com/leonardt/fault/blob/master/.travis.yml#L62-L64

phanrahan, 2019-11-02:
@makaimann, sorry I am so inept. Just can't figure this out. Could we meet between 11 am and 2 pm on Mon?

phanrahan, 2019-11-02:
I was able to install mathsat.

test_constrained_random is still failing. pysmt uses z3-solver 4.8.4. Will that work?

makaimann, 2019-11-02:
Oh no, not inept at all — this installation process is always a pain. We’ve been working on alternatives in CoSA for that very reason.

That’s great you got it installed! It shouldn’t really matter which solver pysmt uses. This is for constrained random right? If it were for CoSA, the only difference would be that we couldn’t run “interpolant-based” model checking and might not be able to prove some properties that we could with mathsat.

phanrahan, 2019-11-04:
Cleaned up all my installs and reinstalled. Seems to work now.

closed 2019-11-04 by phanrahan

leonardt, 2020-01-10:
For future reference, I was running into some issues installing mathsat today.  I was getting this error:
```
   clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]
    ld: library not found for -lstdc++
```

I found this thread: https://stackoverflow.com/questions/56083725/macos-build-issues-lstdc-not-found-while-building-python-package

and the solution: `export MACOSX_DEPLOYMENT_TARGET=10.9` works for me.  It seems that the build flow of mathsat needs to be updated to specify `libc++` instead of `libstdc++` but setting that environment variable works around the problem for now