phanrahan/magma #484: Getting stateless classes working with circuit.sequential

rdaly525, 2019-11-04:
Peak programs need to be written by inheriting from a Peak class. For combinational specs, this works by not implementing the __init__ and only implementing __call__. Is there a current way to compile something like this to a magma circuit? Ideally, I would want the following code to work. This would allow a single way to translate any peak spec to RTL.

```
m.circuit.sequential
class Foo:
  def __call__(self, a: m.Bit) -> m.Bit:
    return ~a
```

rdaly525, 2019-11-04:
One possibility on my end is just applying `circuit.combinational` to the `__call__` method, but this does not work due to the extraneous 'self' argument.

leonardt, 2019-11-04:
This should be simple to support, I will look into it.  It can just pass through the __call__ method to combinational and skip the register logic, i'm guessing there's some code paths that assume a non-zero amount of registers but there's no reason that they shouldn't be able to be changed to support this case.

rdaly525, 2019-11-06:
The difficulty here is dealing with the fact that `__call__` has the first argument as "self" instead of a magma-annotated type.

leonardt, 2019-11-08:
This has been fixed on the `hwtypes` branch by 5450e99, let me know if you need this in master.

rdaly525, 2020-01-23:
@leonardt Reopening this as it is still not working. I added a failing test in test_sequential in the branch 'no-init-test'

commit 1976ad2f (2020-01-23): Merge pull request #542 from phanrahan/no-init-test

Fix #484
closed 2020-01-23 by rdaly525