phanrahan/magma #488: Simple adt test fails with circuit.sequential

rdaly525, 2019-11-08:
Reproducible by checking out the 'hwtypes-product-bug' branch and running:
`>pytest tests/test_syntax/test_sequential.py -k test_product`

Let me know if this is an invalid test.

rdaly525, 2019-11-08:
The error I see is:
`E           Exception: Got None for port 'Foo_comb.Mux2_inst0.I1', is it connected to anything?`

leonardt, 2019-11-08:
pushed fixes to your branch, but it fails the gold check, can you see if the compiler output si good and merge it into hwtypes?

I had to set cache=True (magma needs this for checking type equality, perhaps we can relax this by adding structural checking in the magma internals)

But the main issue was that you were referring to `s` without `self.`

```diff
diff --git a/tests/test_syntax/test_sequential.py b/tests/test_syntax/test_sequential.py
index 5f05bf7..ce41913 100644
--- a/tests/test_syntax/test_sequential.py
+++ b/tests/test_syntax/test_sequential.py
@@ -221,7 +221,7 @@ def test_seq_simple(target, async_reset):
 def test_product(target, async_reset):
     Data = m.Bits[16]

-    class Instr(m.Product):
+    class Instr(m.Product, cache=True):
         a=m.Bit
         b=Data

@@ -234,7 +234,7 @@ def test_product(target, async_reset):
             b = instr.b
             self.s = ~self.s
             if a:
-                return b, s[0]
+                return b, self.s[0]
             else:
                 return ~b, ~a
```

closed 2019-11-08 by leonardt

rdaly525, 2019-11-08:
Ah sorry, that should have been obvious. Maybe a good feature request would be to throw an error earlier if there are references to undefined free variables.

leonardt, 2019-11-08:
yea, it's funky that it actually worked, I think it's because it just moves the code to combinational inside a circuit def where `s` is defined, so it was doing something wonky

leonardt, 2019-11-08:
(worked as in failed later on in the compiler)

leonardt, 2019-11-08:
It's a bit hard to do this check because there could be an `s` (e.g. constant) in the enclosing environment, but let me think if I can come up with a rule that would catch this

leonardt, 2019-11-08:
Here's some valid python code
```
>>> def f():
...     return s
...
>>> s = 3
>>> f()
3
```