phanrahan/magma #491: [feature proposal] Inline syntax for sequential/combinational

leonardt, 2019-11-12:
We've gotten requests for the ability to inline sequential/combinational into a magma circuit definition (similar to how always_ff/comb can reference the current scope of module).  This is convenient for users since there are cases when they want to write a some comb/seq logic and want to avoid the overhead of having to write a separate function/class definition as well as having to explicitly instance/wire the IO.

I think semantically this is fairly simple to add, it basically extends these syntaxes with the ability to reference values in the enclosing scope, however there's some design space for the actual syntax for these code blocks, and we'll need to make sure we clearly define the rules for what values can be operated on in the blocks.

Here's some example code that uses `with` to mark code blocks in a different syntax (we would then do an AST rewrite pass on `definition` using the metaclass hook.
```python
class Foo(m.Circuit):
    IO = [...]

    @classmethod
    def definition(io):
        regs = [...]

        x = m.Bits[2]()
        z = Foo()(x)

        # writing to an anonymous value inside 
        # a sequential block to define a register
        with m.sequential(m.posedge(io.CLK),
                          m.posedge(io.RESET)) as syntax:
            if io.RESET:
                x = 0
            else:
                if regs[0].O:
                    x = 1
                else:
                    x = 2

        # read output values, write to namespace, write to input values
        with m.combinational as syntax:
            if regs[0].O:
                y = 0
                regs[0].I = 4
            else:
                y = io.I
                regs[0].I = 3

        io.O <= y
```

I think as a first pass we can lift these syntax blocks into separate combinational/sequential definitions and insert instantiations and wire up the IO to reflect the expected scope values.  The next step would be have the verilog code generation inline these two separate modules into the current definition so it better reflects the user input.

I'd be interested in any comments on the syntax/semantics.  There's probably some danger in mixing these syntaxes in the same scope, so we'll want to be sure we can clearly define what's legal and what isn't.

phanrahan, 2019-11-12:
It seems good to allow combinational and sequential in definitions. But I am not sure I understand all the semantics issues. Can we just wrap definition in a m.sequential and have it work?

leonardt, 2019-11-12:
I think this may be a better approach, rather than mixing languages in the same scope, we could instead allow the use of core magma inside sequential definitions.  I will draft a proposal of what this could look like.

leonardt, 2020-08-12:
I believe sequential2 (arbitrary code inside __init__) resolves the initial request here, we can open more specific issues.

closed 2020-08-12 by leonardt