phanrahan/magma #494: Sanitize instance/wire/module names

jameshegarty, 2019-11-14:
Currently you can name instances things like “reg”, “wire” etc, which will then emit these names into the verilog file, which will cause some verilog compilers to fail. Can we add a sanity check to any user-entered string to make sure it isn’t a verilog keyword or contains tokens not allowed by verilog?

Btw my preference would be to show an error in these cases (not just silently fix the issue). From my experience it can be annoying if the compiler is changing your variable names behind your back. We should provide a “sanitize” function that the user can explicitly call to clean up their strings.

leonardt, 2019-11-14:
Seems reasonable to me

phanrahan, 2019-11-14:
Seems like a good idea to me too.