phanrahan/magma #495: [rfc] Using @ or @= for wiring instead of <=

leonardt, 2019-11-15:
There's an annoying issue with precedence for `<=`.

When writing `io.a <= b == c`, you have to write `io.a <= (b == c)`.

Furthermore, it's an overloaded operator (means two different things for different types of values), which can lead to confusion.

We're interested in adding support for "wiring" into hwtypes, but @cdonovick has requested that we use `@` or `@=` as the operator, as it makes some analyses easier (don't need to reason about the in/out qualifiers on types).  Furthermore, `@` and `@=` currently don't have any use in HDLs, so it's unlikely to be confusing.  Finally, it's new in Python, so it doesn't have much baggage in terms of user expectations.

Any comments? We can gracefully deprecate `<=` so we don't break existing code if we chose to do this.  I think the one major concern is `@` seems a bit less elegant/familiar than `<=` (non-blocking assignment).

phanrahan, 2019-11-15:
Using `@=` makes a lot of sense to me. We discussed that, but I don't remember what we concluded.

I think the precedence o `@=` is better too. Right?

cdonovick, 2019-11-15:
Related https://github.com/leonardt/hwtypes/issues/102

cdonovick, 2019-11-15:
If we make this change I would be happy to write a refactoring tool to automatically upgrade any legacy magma code to use `@=` instead of `<=` if we okay this change.  I think the other option would be to use `__setitem__` for wiring `x <= y` becomes `x[:] = y` .

closed 2020-01-29 by cdonovick