phanrahan/magma #499: Inconsistency with single bit constants

rdaly525, 2019-11-19:
I would expect the following code to work. (on branch hwtypes)

```
val = m.Bits[4](0)
assert isinstance(val, m.Bits[4]) #This works

val = m.Bit(0)
assert isinstance(val, m.Bit) #This does not work
assert isinstance
```

Why is a single bit constant a 'wire'? Why is this different than a constant Bits?
Given a constant value, is there a consistent way to get its underlying type? (m.Bit for single bit, and m.Bits[n] for multi-bits)

phanrahan, 2019-11-20:
Not sure why that is. I would say it is a bug.

leonardt, 2019-11-20:
Is this on the hwtypes branch? This is likely an issue with how VCC/GND are represented (which is returned by Bit(0)).  Agreed this is a bug, but I think the plan is going to be to remove the wire construct and return to the old treatment of values

rdaly525, 2019-11-20:
Yes, this is on the hwtypes branch. Bit(0) does indeed return GND

phanrahan, 2019-11-20:
We should make sure isinstance(VCC/GND, Bit) is True.

leonardt, 2020-01-14:
As of latest master, `VCC` and `GND` are instances of `Digital`, but not `Bit`, however we can make `isinstance(Bit(0), Bit)` and `isinstance(Bit(1), Bit)`, but they will be different than `VCC` and `GND` (we could name them something like `Bit_True` and `Bit_False` or LOW/HIGH.

Alternatively we can revert VCC/GND to be instances of `Bit` and use something else (e.g LOW/HIGH for Digital).

This is simple to implement so it's mainly a matter of choosing the desired name.

leonardt, 2020-01-14:
Actually, perhaps the simplest current fix is to have VCC be an instance of `Bit` (and therefor an instance of Digital) so it preserves the old naming, and we can decide on how to name 0/1 for digital later.

closed 2020-01-15 by leonardt

phanrahan, 2020-01-15:
Thinking about this in hindsight, I think we should depreciate VCC and GND for values of Bit.

leonardt, 2020-01-15:
That makes sense to me, there is a bit of logic in the internals that compare VCC/GND to represent 0/1, but instead we can have VCC/GND just be names for the `Bit(0)` and `Bit(1)`