phanrahan/magma #506: pytest

phanrahan, 2019-12-09:
I just checked out the latest version. When I run pytest, only 3 tests are run. Is this expected?

@leonardt

leonardt, 2019-12-09:
I think this is related to @rsetaluri latest change w.r.t. to pytest's auto discovery for codestyle (see https://github.com/phanrahan/magma/blob/master/setup.cfg#L8-L9)

For now, you can run `pytest tests` to run the tests explicitly.

@rsetaluri do you know if we can retain the default behavior for test discovery (but not style checks?)

rsetaluri, 2019-12-10:
My understanding of the desired behavior is that for both:
- `pytest tests`
- `pytest`

We effectively want to call (a) `pytest tests/` and (b) `pycodestyle magma/`. Think #508 achieves this.

phanrahan, 2019-12-10:
@rsetaluri , Is this a question? In most repos, pytest runs all the test.

rsetaluri, 2019-12-10:
Less a question, more just suggesting that whatever dev's run should run tests *and* run the style-checks that will be run in CI. Since they may run `pytest` and/or `pytest tests`, both should do the above. This change makes sure of that.

leonardt, 2019-12-10:
With that PR merged running `pytest` now runs all the tests again (but also runs style checks on the package source as well).

closed 2019-12-10 by leonardt