phanrahan/magma #513: Cannot parse Verilog integer literal

Kuree, 2019-12-10:
Error:
```
    def get_value(v, param_map):
        if isinstance(v, pyverilog_ast.IntConst):
>           return int(v.value)
E           ValueError: invalid literal for int() with base 10: "32'h0"
```

SV file:
```
module mod #(parameter KRATOS_INSTANCE_ID = 32'h0)
(
);

endmodule   // mod
```

leonardt, 2019-12-10:
I think we'll need to parse the integer literal string.
 
We could use the size specifier to construct a BitVector of a specific width, read the radix specifier (binary, octal, decimal, hex) to interpret the literal number, read the signed specifier (or lack of it) to decide on UIntVector vs SIntVector interpretation.   I believe coreir only supports unsigned bit vector as a parameter, so we'd want to convert the signed value into an unsigned value (or add support for this in coreir)

leonardt, 2019-12-10:
Alternatively we could find a way to just pass the literal string down the toolchain if the usecase is just to round trip back to verilog.

closed 2019-12-11 by leonardt