phanrahan/magma #527: Lots of errors when running tests

phanrahan, 2020-01-11:
I just checked out the latest version and it does not pass the tests. I also updated hwtypes and ast_tools.
```
tests/test_syntax/test_sequential.py ..FF.F.F..FF..FF.F.F.               [ 42%]
tests/test_syntax/test_to_verilog.py ..                                  [ 42%]
tests/test_type/test_anon.py ..                                          [ 43%]
tests/test_type/test_array.py .......                                    [ 44%]
tests/test_type/test_bit.py ..................                           [ 49%]
tests/test_type/test_bits.py ......................FF......              [ 56%]
tests/test_type/test_clock.py ...............                            [ 59%]
tests/test_type/test_conversions.py .....................ss..            [ 65%]
tests/test_type/test_enum.py ..                                          [ 66%]
tests/test_type/test_new_types.py ..                                     [ 66%]
tests/test_type/test_pretty_print.py ....                                [ 67%]
tests/test_type/test_reset.py .                                          [ 67%]
tests/test_type/test_sint.py ..........................FF..              [ 75%]
tests/test_type/test_tuple.py ........                                   [ 76%]
tests/test_type/test_type_errors.py .....                                [ 78%]
tests/test_type/test_uint.py ........................FF                  [ 84%]
tests/test_verilog/test_declare_params.py .                              [ 84%]
tests/test_verilog/test_ff.py ......                                     [ 85%]
tests/test_verilog/test_from_file.py .....F........                      [ 89%]
```

splhack, 2020-01-11:
Got same errors. I double checked it happens with fresh Ubuntu 18.04 image.
looks like all signed integer has "signed" prefix.

leonardt, 2020-01-11:
Have you updated your versions of coreir/pycoreir? (Can you post the versions you have installed just to double check)

leonardt, 2020-01-11:
Also including more of the error log if there is any would be helpful, as I can't determine anything from that output

leonardt, 2020-01-11:
@splhack can you give me more info on your environment as well? The travis builds are passing so there must be some mismatch but I could be wrong: https://travis-ci.org/phanrahan/magma/jobs/635551827

leonardt, 2020-01-11:
And can you expand on what you mean by
> like all signed integer has "signed" prefix.

leonardt, 2020-01-11:
I was able to reproduce this on an ubuntu VM, it looks like the linux wheel for pycorier is out of date with the coreir master branch, will try doing a new release.

It works on Mac though, so @phanrahan I suspect you are using an older version of pycoreir/coreir (the error I was seeing was a missing `1'b` prefix on boolean constants in the generated verilog, which was added in the latest version of coreir.

leonardt, 2020-01-11:
Actually, it looks like I was able to pass the test suite with the existing coreir/pycorier release. @splhack can you run `pip show coreir` to show what version you are using? And are you using your own installation of coreir? Or are you using the distributed binary wheel?

Also, can you provide exactly which tests are failing and the message?

leonardt, 2020-01-11:
Also, the steps you used to reproduce the issue on the fresh VM would be helpful

splhack, 2020-01-11:
@leonardt here is my all-github-master with `pip3 install -e .` env.

But I also tested fresh Ubuntu 18.04 image, and `pip3 install -e .` from git-clone-ed magma. It means mantle, pycoreir, coreir, etc all packages installed by pip3. and same result.

- coreir
  - master 5279e70fbdec72060d58c99c1b618fc19b208fcf
  - cmake -DCMAKE_INSTALL_PREFIX=/opt/FPGA; make -j$(nproc); suto make install
  - it would be nice if coreir has `--version` option

    $ which coreir
    /opt/FPGA/bin/coreir
    $ coreir --version
    terminate called after throwing an instance of 'cxxopts::option_not_exists_exception'
      what(): Option ‘version’ does not exist

- verilogAST (just in case, double checked automatically cloned into coreir/verilogAST-src)
  - master 149347b8aa016c195f3375052b353a9132e0dc38
- pycoreir
  - master 41f3b3d0ff634ff3f08cc27e98d948a670877435
- mantle
  - master c51c74cc61d26a7013aaff05f2328f861ba8552d
- magma
  - master 6d6a9d06d78e30250aee43038ab29d7ac0203e40

- pip list
    coreir (2.0.48, /SOMESRCPATH/pycoreir)
    fault (3.0.0, /SOMESRCPATH/fault)
    hwtypes (1.3.4, /SOMESRCPATH/hwtypes)
    loam (0.1a0, /SOMESRCPATH/loam)
    magma-lang (2.0.2, /SOMESRCPATH/magma)
    mantle (2.0.0, /SOMESRCPATH/mantle)

- most of the pytest errors is due to `signed` mismatch between gold vs build

    ----------------------------- Captured stdout call -----------------------------
    /SOMESRCPATH/magma/tests/test_wire/build/named2c.v /SOMESRCPATH/magma/tests/test_wire/gold/named2c.v
    ----------------------------- Captured stderr call -----------------------------
    --- gold/named2c.v
    +++ build/named2c.v
    @@ -1,4 +1,4 @@
    -module main (input [1:0] I, output  O);
    +module main (input signed [1:0] I, output  O);
     wire  And2_inst0_O;
     And2 And2_inst0 (.I0(I[0]), .I1(I[1]), .O(And2_inst0_O));
     assign O = And2_inst0_O;

leonardt, 2020-01-11:
thanks, I'll try to reproduce this, it's a very strange error.  That error looks related to the verilog backend code generation since it's part of the `test_named4` which doesn't use coreir

leonardt, 2020-01-12:
Hmm, I was unsuccessful at reproducing this on a fresh ubuntu 18.04 image (cloning magma master, pip install from source)

leonardt, 2020-01-12:
I did use Miniconda for python, so maybe this has to do with ubuntu python? I'll try that

leonardt, 2020-01-12:
Ah, @splhack are you using python3.6 (seems to be the case for default python3 for me) on your fresh image? The pycoreir package distributes a wheel for python3.7, so it's likely that you would need to compile coreir from source in the pip install step.  I'm still waiting for that to finish, but it could perhaps be the case that there's some mismatch between the python3.6/python3.7 behavior (I've only tested 3.7).  I will see what happens when the build is done though.

leonardt, 2020-01-12:
Ah, I was able to reproduce issues with `sudo apt-get install python3.7`, investigating this further, it seems that using miniconda3 works, so there might be some incompatibility with the python versions.

leonardt, 2020-01-12:
Hmm my error seemed to be a missing coreir script in my path, so back to square one (tests passing on python3.7 ubuntu 18.04)

splhack, 2020-01-12:
I found a different issue (missing shared library file) on pycoreir when I tried to repro the issue on Ubuntu 18.04. will try with Python 3.7.

    $ python3 --version
    Python 3.6.9
    $ sudo apt install python3-pip
    $ git clone https://github.com/phanrahan/magma.git
    $ cd magma
    $ pip3 install -e .
    $ pip3 install pytest
    $ pip3 list
    ...
    coreir (2.0.49)
    ...
    magma-lang (2.0.2, /home/kaz/magma)
    ...
    $ export PATH=$HOME/.local/bin:$PATH
    $ export LD_LIBRARY_PATH=$HOME/.local/lib/python3.6/site-packages/coreir
    $ which coreir
    /home/kaz/.local/bin/coreir
    $ coreir
    ...
    OSError: libcoreirsim.so: cannot open shared object file: No such file or directory
    $ ls $HOME/.local/lib/python3.6/site-packages/coreir/*.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-aetherlinglib.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-c.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-commonlib.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-float.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-float_CW.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-float_DW.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-ice40.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreir-rtlil.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libcoreirsim-c.so
    /home/kaz/.local/lib/python3.6/site-packages/coreir/libverilogAST.so

splhack, 2020-01-12:
Python 3.7 works! (test failed due to missing verilator, because it's fresh image)

    $ sudo apt install python3.7 python3.7-dev python3-pip
    $ sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 1
    $ python3 --version
    Python 3.7.5
    $ pip3 --version
    pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.7)
    $ git clone https://github.com/phanrahan/magma.git
    $ cd magma
    $ pip3 install -e .
    $ pip3 install pytest
    $ pip3 list
    ...
    coreir (2.0.49)
    ...
    magma-lang (2.0.2, /home/kaz/magma)
    ...
    $ export PATH=$HOME/.local/bin:$PATH
    $ which coreir
    /home/kaz/.local/bin/coreir
    $ coreir
    ...
    /pycoreir/coreir-cpp/src/ir/passmanager.cpp:248   wireclocks-coreir
    $ pytest
    ...
    ============= 2 failed, 411 passed, 7 skipped in 66.25s (0:01:06) ==============
    ...
    /bin/sh: 1: verilator: not found

splhack, 2020-01-12:
verified all-github-master env (https://github.com/phanrahan/magma/issues/527#issuecomment-573362526) with self-built-coreir works with Python 3.7.

    ================================= 413 passed, 7 skipped in 22.62s ==================================

phanrahan, 2020-01-12:
I had forgotten to update coreir and pycoreir. Once I updated, the tests pasts. Sorry for fire alarm.

closed 2020-01-12 by leonardt