phanrahan/magma #529: inline=True doesn't generate `corebit_*` to Verilog

splhack, 2020-01-12:
Same versions as #528
Add `inline=True` to test_bit.py

    --- a/tests/test_type/test_bit.py
    +++ b/tests/test_type/test_bit.py
    @@ -281,7 +281,7 @@ wire(TestBinary.I1, magma_Bit_{clean_op}_inst0.in1)
     wire(magma_Bit_{clean_op}_inst0.out, TestBinary.O)
     EndCircuit()\
     """
    -    m.compile(f"build/TestBit{clean_op}", TestBinary, output="coreir-verilog")
    +    m.compile(f"build/TestBit{clean_op}", TestBinary, inline=True, output="coreir-verilog")
         assert check_files_equal(__file__, f"build/TestBit{clean_op}.v",
                                  f"gold/TestBit{clean_op}.v")

The generated Verilog doesn't have `corebit_*` modules. Example,

    --- gold/TestBitxor.v
    +++ build/TestBitxor.v
    @@ -1,10 +1,4 @@
    -module corebit_xor (input in0, input in1, output out);
    -  assign out = in0 ^ in1;
    +module TestBinary (input I0, input I1, output O);
    +corebit_xor magma_Bit_xor_inst0(.in0(I0), .in1(I1), .out(O));
     endmodule

    -module TestBinary (input I0, input I1, output O);
    -wire magma_Bit_xor_inst0_out;
    -corebit_xor magma_Bit_xor_inst0(.in0(I0), .in1(I1), .out(magma_Bit_xor_inst0_out));
    -assign O = magma_Bit_xor_inst0_out;
    -endmodule
    -

Am I missing something?

leonardt, 2020-01-12:
Thanks, this looks like a bug, will investigate.

leonardt, 2020-01-12:
(I believe that it's likely omitting the module because it should be inlined, but it's not actually doing the inlining at the instantiation site)

leonardt, 2020-01-13:
Fix submitted by https://github.com/rdaly525/coreir/pull/827

leonardt, 2020-01-13:
Should be fixed by pycoreir==2.0.50 once it's released, let me know if that's not the case.

closed 2020-01-13 by leonardt