phanrahan/magma #535: Inline issue

splhack, 2020-01-14:
for example, taken from 
 https://github.com/phanrahan/magma/blob/master/tests/test_syntax/test_sequential.py#L306-L343

    import magma as m

    async_reset = False
    T=m.Bits[4]

    @m.circuit.sequential(async_reset=async_reset)
    class Register:
        def __init__(self):
            self.value: T = T(0)

        def __call__(self, value: T, en: m.Bit) -> T:
            retvalue = self.value
            if en:
                self.value = value
            else:
                self.value = self.value
            return retvalue

    @m.circuit.sequential(async_reset=async_reset)
    class RegisterMode:
        def __init__(self):
            self.register: Register = Register(0)

        def __call__(self, mode: m.Bits[2], const_: T, value: T, clk_en: m.Bit,
                     config_we: m.Bit, config_data: T) -> (T, T):
            if config_we == m.Bit(1):
                reg_val = self.register(config_data, m.Bit(1))
                return reg_val, reg_val
            elif mode == m.bits(0, 2):
                reg_val = self.register(value, m.Bit(False))
                return const_, reg_val
            elif mode == m.bits(1, 2):
                reg_val = self.register(value, m.Bit(False))
                return value, reg_val
            elif mode == m.bits(2, 2):
                reg_val = self.register(value, clk_en)
                return reg_val, reg_val

    m.compile("build/top", RegisterMode, output="coreir-verilog", inline=True)

Generated Verilog has missing connection.

    yosys -q -p 'synth_ice40 -top RegisterMode' build/top.v
    Warning: Wire RegisterMode.\RegisterMode_comb_inst0.Mux2xOutBits4_inst13.coreir_commonlib_mux2x4_inst0.in_data_1 [3] is used but has no driver.
    Warning: Wire RegisterMode.\RegisterMode_comb_inst0.Mux2xOutBits4_inst13.coreir_commonlib_mux2x4_inst0.in_data_1 [2] is used but has no driver.
    Warning: Wire RegisterMode.\RegisterMode_comb_inst0.Mux2xOutBits4_inst13.coreir_commonlib_mux2x4_inst0.in_data_1 [1] is used but has no driver.
    Warning: Wire RegisterMode.\RegisterMode_comb_inst0.Mux2xOutBits4_inst13.coreir_commonlib_mux2x4_inst0.in_data_1 [0] is used but has no driver.

looks like `self_value_O` isn't connected?

     module Register_comb (output [3:0] O0, output [3:0] O1, input en, input [3:0] self_value_O, input [3:0] value);
    -wire [3:0] Mux2xOutBits4_inst0_O;
    -Mux2xOutBits4 Mux2xOutBits4_inst0(.I0(self_value_O), .I1(value), .O(Mux2xOutBits4_inst0_O), .S(en));
    -assign O0 = Mux2xOutBits4_inst0_O;
    -assign O1 = self_value_O;
    +Mux2xOutBits4 Mux2xOutBits4_inst0(.I0(O1), .I1(value), .O(O0), .S(en));
     endmodule

leonardt, 2020-01-14:
Looks like a bug will investigate

cdonovick, 2020-01-14:
Possibly(?) related to:
https://github.com/phanrahan/magma/issues/430

leonardt, 2020-01-14:
Fixed by https://github.com/leonardt/verilogAST-cpp/pull/36, the issue was the output assign inline logic was being performed in a case when it shouldn't.  Normally, if an output is assigned to a value that is not assigned (e.g. used as a module instance output) we can replace that value with the output (to effectively act as the wire).  However, in this case, the value was an input, which should not be "inlined".  The above PR adds a guard that prevents the output assign inline logic from being run when the output is being assigned to an input.

splhack, 2020-01-15:
verified https://github.com/leonardt/verilogAST-cpp/pull/36 fixed missing connections on my design too!

closed 2020-01-15 by leonardt