phanrahan/magma #543: inline=True + verilator_debug=True issue

splhack, 2020-01-27:
Repro code

    import magma as m

    class Out(m.Product):
        A = m.Bit

    @m.circuit.sequential(async_reset=False)
    class Top:
        def __call__(self) -> Out:
            return m.namedtuple(A=m.bit(0))

    m.compile("build/top", Top, output="coreir-verilog", inline=True, verilator_debug=True)

In generated Verilog, `O_A` wasn't connected.

    module Top_comb (output O_A/*verilator public*/);
    endmodule

    module Top (input CLK/*verilator public*/, output O_A/*verilator public*/);
    wire Top_comb_inst0_O_A;
    Top_comb Top_comb_inst0(.O_A(Top_comb_inst0_O_A));
    endmodule

`inline=True, verilator_debug=False` works

    Top_comb Top_comb_inst0(.O_A(O_A));

And `inline=False, verilator_debug=True` too

    wire Top_comb_inst0_O_A;
    Top_comb Top_comb_inst0(.O_A(Top_comb_inst0_O_A));
    assign O_A = Top_comb_inst0_O_A;

leonardt, 2020-01-29:
This is likely because we hacked support for Verilator public into the verilog backend by using an inline string, so the inlining logic isn't picking up the fact that O_A is an output port.  Will fix

closed 2020-02-05 by leonardt

leonardt, 2020-02-05:
Should be fixed in newest pycoreir release (2.0.55)