phanrahan/magma #547: New wire syntax and arrays

cdonovick, 2020-01-28:
Was updating magma code to use the new wire syntax when I discovered a problem, wiring arrays elements doesn't work.
Consider:

```Python
arr = Array[N, T]()
arr[0] @= elem
```
Which is equivalent to:
```Python
temp = arr[0] # arr.__getitem__(0)
temp @= elem # temp.__imatmul__(elem)
arr[0] = temp # arr.__setitem__(0, temp) <--- THE PROBLEM
```

Now the old syntax worked because:
```Python
arr[0] <= elem
```
is equivalent to:
```Python
temp = arr[0] # arr.__getitem__(0)
temp <= elem # temp.__le__(elem)
```

I think I have pretty simple solution.

Currently `t.__imatmul__` returns `self`, which is probably just a stray piece of code it should logicall return `None`.  However, this is perfect and allows the following solution:

```Python
class Array(...):
    ...
    def __setitem__(self, key, val):
        old = self[key]
        if old is not val:
            raise ValueError()
```

This wouldn't change much about allowable syntax beyond permitting tautological statements like:
```Python
arr[0] = arr[0]
```

leonardt, 2020-01-29:
I think this is reasonable, since array doesn't define `__setitem__` currently, and in fact, we probably want to discourage mutating arrays, which this effectively prevents

leonardt, 2020-01-29:
It also should catch bugs where someone tries to wire an array with `arr[0] = driver` and force them to use `@=` for wiring instead, which should avoid a class a bugs.

leonardt, 2020-01-30:
This issue also occurs with tuples, e.g.
```
            class RTL(m.Circuit):
                IO = ["CLK", m.In(m.Clock),
                      "in1", m.In(m.Bits[width]),
                      "in2", m.In(m.Bits[width]),
                      "out", m.Out(m.Bit),
                      "handshake", HandShake]

                @classmethod
                def definition(cls):
                    temp1 = orr()(cls.in1)
                    temp2 = andr()(cls.in1)
                    cls.out @= logical_and()(temp1, temp2)
                    cls.handshake.valid @= cls.handshake.ready

```

produces

```
    cls.handshake.valid @= cls.handshake.ready
../../miniconda3/lib/python3.7/site-packages/hwtypes/util.py:108: in __set__
    self.fset(obj, value)
magma/tuple.py:318: in prop
    self[idx] = value
../../miniconda3/lib/python3.7/site-packages/hwtypes/adt.py:47: in __setitem__
    if isinstance(value, type(self)[idx]):
../../miniconda3/lib/python3.7/site-packages/hwtypes/adt_meta.py:389: in __getitem__
    return super().__getitem__(idx)
```

Would the fix be similar?

leonardt, 2020-01-30:
Handshake type definition
```
            class HandShake(m.Product):
                ready = m.In(m.Bit)
                valid = m.Out(m.Bit)
```

cdonovick, 2020-01-30:
@leonardt could you post the full stack trace?  What was the actual error?
In any event.  Probably.

leonardt, 2020-01-30:
Minimal repro:

```python
# tmp.py
import magma as m

class HandShake(m.Product):
    ready = m.In(m.Bit)
    valid = m.Out(m.Bit)

width = 5

class RTL(m.Circuit):
    IO = ["handshake", HandShake]

    @classmethod
    def definition(cls):
        cls.handshake.valid @= cls.handshake.ready
```

```
❯ python tmp.py
/Users/leonardtruong/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning:
Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it
 will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
Traceback (most recent call last):
  File "tmp.py", line 12, in <module>
    class RTL(m.Circuit):
  File "/Users/leonardtruong/repos/magma/magma/circuit.py", line 458, in __new__
    self.definition()
  File "tmp.py", line 17, in definition
    cls.handshake.valid @= cls.handshake.ready
  File "/Users/leonardtruong/miniconda3/lib/python3.7/site-packages/hwtypes/util.py", line 108, in __set__
    self.fset(obj, value)
  File "/Users/leonardtruong/repos/magma/magma/tuple.py", line 318, in prop
    self[idx] = value
  File "/Users/leonardtruong/miniconda3/lib/python3.7/site-packages/hwtypes/adt.py", line 47, in __setitem__
    if isinstance(value, type(self)[idx]):
  File "/Users/leonardtruong/miniconda3/lib/python3.7/site-packages/hwtypes/adt_meta.py", line 389, in __geti
tem__
    return super().__getitem__(idx)
  File "/Users/leonardtruong/miniconda3/lib/python3.7/site-packages/hwtypes/adt_meta.py", line 292, in __geti
tem__
    return cls.fields[idx]
TypeError: 'odict_values' object does not support indexing
```

cdonovick, 2020-01-30:
I'll investigate this.

leonardt, 2020-02-04:
Re opening because it was automatically closed by #570 but I think there's still the array issue?

leonardt, 2020-02-04:
Oh wait that was resolved in https://github.com/phanrahan/magma/pull/552/files

@cdonovick is there anything left in this one?

cdonovick, 2020-02-12:
@leonardt 
Not that I am aware of. I had intentionally let #570 close this.
The only other place where the new syntax creates issue that I am aware of is:
```Python
getattr(foo, field) <= value # works
getattr(foo, field) @= value # SyntaxError
```
But as the issue is a `SyntaxError` there is no hope of fixing it.  We can just tell people to use explicit wire calls if they they need to `getattr`

PR #570 (2020-02-04): Fix wiring with products
resolves #547 part b
closed 2020-02-12 by leonardt