phanrahan/magma #555: circuit.sequential fails on calls within branches

rdaly525, 2020-01-30:
```
    @m.circuit.sequential
    class A:
        def __call__(self, i: m.Bit) -> m.Bit:
            return i

    #Works 
    @m.circuit.sequential
    class B0:
        def __init__(self):
            self.a0: A = A()
            self.a1: A = A()

        def __call__(self, i : m.Bit) -> m.Bit:
            i0 = self.a0(i)
            i1 = self.a1(~i)
            if i:
                return i0
            else:
                return i1

    #Currently fails
    @m.circuit.sequential
    class B1:
        def __init__(self):
            self.a0: A = A()
            self.a1: A = A()

        def __call__(self, i : m.Bit) -> m.Bit:
            if i:
                return self.a0(i)
            else:
                return self.a1(~i)
```

I have a branch called 'hierarchy' which includes this test in test_sequential.py

cdonovick, 2020-01-30:
Did you push? All tests pass on your branch

cdonovick, 2020-01-30:
Actually never mind I have no problem recreating the issue