phanrahan/magma #559: Can sub-portion of Product be set?

splhack, 2020-01-31:
Probably I should use `m.namedtuple` to set `Product` but let me give it a shot.

Isn't `a.a0 = value` the same thing as `a = m.namedtuple(a0=value, a1=a.a1)`?

    import magma as m
    import fault
    
    
    class A(m.Product):
        a0 = m.Bits[8]
        a1 = m.Bits[8]
    
    
    @m.circuit.sequential(async_reset=False)
    class TestTupleUpdate:
        def __init__(self):
            self.a: A = m.namedtuple(a0=0, a1=0)
    
        def __call__(self, sel: m.Bit, value: m.Bits[8]) -> A:
            a = self.a

            # TypeError: 'odict_values' object is not subscriptable    
            # if sel:
            #    a.a0 = value
            # else:
            #    a.a1 = value
    
            if sel:
                a = m.namedtuple(a0=value, a1=a.a1)
            else:
                a = m.namedtuple(a0=a.a0, a1=value)
    
            self.a = a
            return a
    
    
    tester = fault.Tester(TestTupleUpdate, clock=TestTupleUpdate.CLK)
    
    tester.circuit.sel = 0
    tester.circuit.value = 11
    tester.step(2)
    
    tester.circuit.O.a0.expect(0)
    tester.circuit.O.a1.expect(11)
    
    tester.circuit.sel = 1
    tester.circuit.value = 12
    tester.step(2)
    
    tester.circuit.O.a0.expect(12)
    tester.circuit.O.a1.expect(11)
    
    tester.compile_and_run(
        "verilator",
        flags=["-Wno-fatal"],
        magma_opts={"verilator_debug": True},
        directory="build",
    )

leonardt, 2020-01-31:
I think this is related to the comment in this thread https://github.com/phanrahan/magma/issues/547 related to tuples (wiring a field of the tuple is an error but we plan to support it)

leonardt, 2020-01-31:
I think setting partial components of a tuple should be supported if we can.  Basically, it allows you to incrementally (across multiple statements) modify the elements.  The final value of the tuple is then either returned or wired up to the register inputs.

cdonovick, 2020-02-03:
Atleast the type error odict values thing should be resolved by #570

leonardt, 2020-02-04:
The issue is now that it's trying to mutate a tuple field (rather than wire to it).

It's saying, update a field in `self.a`, but keep the old value for other fields, so it's basically an alias for the code written explicitly `a = m.namedtuple(a0=a.a0, a1=value)`.

One option would be to have `__setitem__` return a new copy of the value with the updated field (kind've like immutable arrays).

I think this requires thought as to whether we can support this style of mutation in comb/seq.  For example, would we want to allow setting individual elements of an array?

Now that I think about, calling `__setitem__` on an output value either should be an error (which is what is going on here) or it should return a new copy of the value with the item set (immutable values).  The immutable copy might not be unreasonable, because that's effectively what will happen in hardware (you aren't setting the value of a the tuple, you're creating a new set of wires with a subset of the wires replaced with the new value).

perhaps we could find another syntax for making a partial copy with an updated field/index?

cdonovick, 2020-02-04:
The issue with copy / return idea you are suggesting is that impossible to collect the return value of a `__setitem__` or `__setattr__` at least not in the way anyone would ever write code.

```
class Foo:
  def __setitem__(self, idx, value):
     return 5

foo = Foo()

f[0] = 0 # returns 5 but nobody gets to see it
assert f.__setitem__(0, 0) == 5
```

We could have a method specifically for partial replacements though:
```
class A(m.Product):
    a0 = m.Bits[8]
    a1 = m.Bits[8]

a = A(...)

new_a = a.replace(a0=new_val)
```

leonardt, 2020-02-04:
That seems like a reasonable syntax, and makes the concept of "mutate-by-copy" explicit, which I like.

splhack, 2020-02-04:
sound good to me! do we follow the Python namedtuple method name `_replace`?
https://docs.python.org/3/library/collections.html#collections.somenamedtuple._replace

leonardt, 2020-02-04:
Seems like a reasonable choice (although I wonder why they used an underscore? is replace a keyword?)

splhack, 2020-02-04:
https://docs.python.org/3/library/collections.html#namedtuple-factory-function-for-tuples-with-named-fields
> To prevent conflicts with field names, the method and attribute names start with an underscore

leonardt, 2020-02-04:
Ah makes sense, seems like we should do the same for magma types to help reduce the number of reserved names (w.r.t. to https://github.com/phanrahan/magma/issues/563)

PR #572 (2020-02-26): replace method for Array/Tuple/Product
close #559 

    val0 = m.namedtuple(x=m.bit(0), y=m.bit(1))
    val1 = val0._replace(x=m.bit(1))
closed 2020-02-26 by leonardt