phanrahan/magma #563: Enum namespace

leonardt, 2020-01-31:
We should either:
(a) have a list of reserved names and error when a user provides an enum field that uses one (see https://github.com/phanrahan/magma/issues/500)
(b) use a separate namespace for enum fields that avoids clobbering the internally used names by the type system, so the user can use arbitrary names without breaking internal code.

splhack, 2020-02-02:
Product needs same thing #567

cdonovick, 2020-02-04:
hwtypes uses a list of reserved `names` which may not be fields.
There is also list of reserved `_names_` which if set will have strange side effects but allow for extension.

https://github.com/leonardt/hwtypes/blob/master/hwtypes/adt_meta.py#L40
https://github.com/leonardt/hwtypes/blob/master/tests/test_adt.py#L359

It would make sense for magma to do something