phanrahan/magma #571: Type polymorphism weirdness

rdaly525, 2020-02-04:
I would suspect the following to error in magma but it is falsely working. lassen:master is doing something equivalent to the following:
```
def test_adc():
    Data = m.Bits[16]
    UData = m.UInt[16]
    SData = m.SInt[16]
    @m.circuit.sequential(async_reset=True)
    class ADC:
        def __call__(self, s:m.Bit, a:Data, b:Data) -> (Data, m.Bit):
            if s:
                a = SData(a)
                b = SData(b)
            else:
                a = UData(a)
                b = UData(b)
            return a.adc(b,s)
```

from the ast_tools SSA pass which is more conservative, this fails complaining about 'a' being different types on each branch (inputs to the phi)

I have this test in the branch magmag:adc

cdonovick, 2020-02-05:
Untested but my guess is that magma allows `condition.ite(signed, unsigned)`.  Which hwtypes would reject,  while `signed` and `unsigned` share a common base I do not think it appropriate to operate in such a manner.  I don't want: 
```
class Foo: pass
foo = Foo()
bar = BitVector[8](...)
con = Bit(...)
assert type(con.ite(foo, bar)) is object
```
I want it to error (technically it would error as you can't construct `object` from other of the arguments, but my point should be clear).

Related to https://github.com/leonardt/hwtypes/issues/122

leonardt, 2020-02-05:
Yes this should be an error and it's probably an artifact of the subclass relationship as Caleb pointed out.

leonardt, 2021-01-22:
Should be fixed by #892, let me know if it isn't

closed 2021-01-22 by leonardt