phanrahan/magma #575: [RFC] Naming Temporary Values

leonardt, 2020-02-07:
There's been requests for support for passing named temporary values down through the toolchain.  For now, we could add support for named values such as `x = Bit(name="x")` and find a way to pass that name through the toolchain so that coreir generates a `wire x`.

However, there's also been requests for automatically inserting this name (so supporting `x = Bit()` and automatically capturing the name `x`).  One simple option is to use `inspect` inside the constructor to find the source file/line and retrieve the LHS of the assignment statement.  However, this is problematic from a performance perspective, seeing as it will require a stack frame traversal for every intermediate value that's constructed.  There's further complications when you consider python code such as `x, y = Bit(), Bit()` (which variable should we use?).

Perhaps a more performant and robust approach would be to traverse the AST and annotate names a preprocessing step.  That would cover the tuple assignment case (because we can match on each side and rewrite it to be `x, y = Bit(name="x"), Bit(name="y")`.

But when I think about this more generally, there's challenges handling the arbitrary python code that magma users can write.  For example, consider `x = f(...)` where `f` returns a `Bit()`.  Ideally, we'd be able to capture the name of the Bit as `x` (this also goes for the result of an expression such as `x = y + z`).

One way to perhaps capture this pattern is to add a wrapper function for assignments, so `x = f(...)` would become `x = m._set_name("x", f(...))`.  Then, magma could inspect the result of `f`, if it's a magma value, set the name to be `x`.  

Any thoughts? I'm particularly interested in whether there's any gotchas here, or more clever/robust ways to handle this.  (I don't think calling `inspect` will be a scalable solution).

We could also use the pre-processing the AST step to capture the filename/lineno (avoiding the need for inspect in debug_mode).  Normally rewriting the AST is tricky because it can obscure error messages, but in this case, we can do this by just wrapping assignment values with a function call, so ideally we are not changing any filename/linenumbers (there's some trickiness here with deserializing the transformed AST, because we could lose the original linebreaks/comments which actually could mess up the mapping to the original source, but maybe this is not an issue if we can capture the original file/line which we'll need for the debug info anyways).

cdonovick, 2020-02-07:
;I'd be surprised it there exists some solution that doesn't incur some massive performance penalty.
I'm also not entirely clear on desired semantics.  For example how should the following respond:
```Python
def returns_3_bits():
  return Bit(), Bit(), Bit()

x, *others = returns_3_bits()

lst = [None]
lst[0] = Bit()

y = Bit(name='x')

def returns_implicit_named_bit():
  y = Bit()
  return y

x = returns_implicit_named_bit()

def returns_explicitly_named_bit():
   y = Bit(name='y')

x = returns_explicitly_named_bit()
```

Now assuming we resolve these ambiguous cases , I think it could be done robustly if we used my ast_tools  instrumentation importer or limited ourselves to circuits declared using the class constructor.  I do see how we could ever hope to capture the following without importer support:
```Python
#module.py
import magma as m

main =  m.DefineCircuit("main", "I", m.In(Bit), "O", m,Out(Bit))
x = m.Bit()
m.wire(main.I, x)
m.wire(x, main.O)
m.EndCircuit()
```

Now if we assume that we have access to the AST of all declaration points.  Then we can do something it just seems like it will probably be incredibly costly.  We could find ALL assignment statements and insert the following logic around them as @leonardt  suggested.

```Python
x = Bit()
_set_name_(x, 'x')
```

Now `_set_name` itself is going to have do something along the lines of:
```Python
def _set_name(obj, name):
  if isinstance(obj, m.Type):
    if obj.is_anon(): # assuming we don't want overwrite explicit names
      try:
        obj._set_name(name)
      except:
        pass
```
That's a lot  branches  / calls  for literally every assignment statement.

So I think this is probably a very bad idea.

rsetaluri, 2020-02-07:
We anyways want to move away from/deprecate using `m.DefineCircuit` and only use the class syntax. We also want to get rid of having to do `def definition(io)` and allow doing wiring inline in the class body.

Assuming both of these are done, circuits will now look like:

```python
class Foo(m.Circuit):
    io = IO(I=m.In(m.Bits[2]), O=m.In(m.Bits[2]))
    io.O[0] @= io.I[0]
    x = IO.I[1]
    io.O[1] @= x
```

Now I believe using metaclass `__prepare__` (https://www.python.org/dev/peps/pep-3115/#invoking-the-metaclass), we can inject a special dict-like `namespace` object during class definition. Which means (I think) we could check that every time we try to assign to a temporary, we could give it the name specified. And now that I think of it, all those temporaries are assigned to the class anyway, meaning we could access them after the fact (and avoid `__prepare__` business altogether).

This doesn't solve many of the other issues you bring up Caleb, but since this syntax is the direction we want to go anyway, it seems like we get ^ for free.

@cdonovick

cdonovick, 2020-02-07:
@rsetaluri 

You are right in theory about `__prepare__` as long the only place where things are defined are in the class body.  You should be able define `__setitem__` on object returned from `__prepare__` to basically do the wrapping, hence the costs would only be seen for assignments within the class.   However, any name you create inside a method or function would not have that behavior.  There is no `__prepare__` for functions.  

But alas in practice you would be wrong: https://bugs.python.org/issue17421 

Also if the above wasn't true there would be a number of reasons you would probably want to use `__prepare__` and not try to recreate it after the fact but I don't think that's particularly relevant *right now*.

cdonovick, 2020-02-07:
Also:

> Explicit is better than implicit.

\- The Zen of Python, by Tim Peters

rsetaluri, 2020-02-07:
@cdonovick yeah agree with all. My only contention would be that this (if it worked in practice) would be (a) low perf overhead (no ast) and (b) very well defined (it only captures names inline in the class body, everything else is up to you to do explicitly).

And yeah, I guess with this, no way around the implicit vs explicit

cdonovick, 2020-02-07:
@rsetaluri 
Very well defined until you hit any of my ambiguous cases...  I can recreate all of them except for `returns_implicit_named_bit` inside that syntax. 

It also creates new ambiguous cases (which are the reasons why `__prepare__` would be what you want to use)

```Python
class Foo(m.Circuit):
    io = IO(I=m.In(m.Bits[2]), O=m.In(m.Bits[2]))
    x = io.I[0]
    io.O[0] @= x
    x = io.I[1] # this will be x in the final namespace
    io.O[1] @= x

class Foo(m.Circuit):
    io = IO(I=m.In(m.Bits[2]), O=m.In(m.Bits[2]))
    x = io.I[0]
    io.O[0] @= x
    x = io.I[1]
    io.O[1] @= x
    del x  # x won't be in the final namespace
```

cdonovick, 2020-02-07:
I think a convenience magic with poorly defined rules about how it will be invoked and what the results will be is significantly worse than not having magic at all.

cdonovick, 2020-02-07:
I should add that if all my concerns could be addressed I think this a great idea. There is no denying that `foo = Bit(name='foo')` is annoyingly verbose.