phanrahan/magma #587: Arithmetic operators on Bits

cdonovick, 2020-02-13:
Currently every implementation of `AbstractBitVector` other than `Bits`  supports the use of arithmetic `__operators__` (`__add__`, `__sub__`, etc...).   While not specifically defined in the abc this had lead to a widespread assumption that `AbstractBitVector` supports these operators, further, it leads to PEak code that works in every context except for magma.  

For example the following mantle test assume that `BitVector` defines `__add__`
https://github.com/phanrahan/mantle/blob/master/tests/test_coreir/test_arith.py#L10

Now, I believe that one following must happen: 
  - `Bits` needs to add support for arithmetic ops, 
  -  we need to remove them from `BitVector` so that two are consistent
  -  we need to create a `MagmaBV` distinct from `Bits` behaves like the rest of hwtypes.   

Now removing support for arithmetic in `BitVector` would mean breaking almost every consumer of hwtypes.  Adding them to `Bits`  or creating a `MagmaBV` would break nothing.

cdonovick, 2020-02-13:
Personally I think the easiest would be to add `MagmaBV` with the new magma protocol it would be very easy to get it to work with the rest of the infrastructure.   While unifying the the interfaces between magma and hwtypes.

rdaly525, 2020-02-17:
I believe we should keep all arithmetic operators (along with operations like sext) on hwtypes.BitVector and hwtypes.SMTBitVector as this maintains the QF BV SMT semantics. It is definitely annoying to have magma.Bits behave differently here when using it as a peak compilation target, so creating a consistent MagmaBV seems like a reasonable idea.

PR #792 (2020-08-25): Define arith ops on Bits for consistency with hwtypes
Fixes #587

Should we also define the comparison ops?
commit dc619d5c (2020-08-25): Define arith ops on Bits for consistency with hwtypes

Fixes #587
closed 2020-08-25 by leonardt