phanrahan/magma #596: Clock interface support for m.IO and incremental construction

leonardt, 2020-02-25:
We used to do `IO += m.ClockInterface()` and it would just use list extension, is there a nice pattern we can use for the new `IO` object?

It also strikes me that it would be useful to incrementally build the IO object, for example
```
io = m.IO()
for i in range(10):
    io[f"I{i}"] = m.In(m.Bit)
```

could we support something like this?

For the clock interface pattern, we could overload `__add__` and support the old list pattern (for backwards compatibility)

rsetaluri, 2020-02-26:
We definitely could support these. Let's use this issue to come to a consensus on syntax.

I think a cool thing to do would be to be able to "add" IO's together (ala bundle's). This would merge the names, and to start with we could say any duplicate names are not allowed. So

```python
x = m.IO(...)
y = m.IO(...)

io = x + y  # errors if x and y share any names
# do some wiring stuff...
io += m.IO(...)
# do some more wiring stuff w/ new ports...
```
An extension could be to merge common names if they have the same type, and error otherwise.

Specifically I would advocate for the following:
* Overloading `+` and `+=` methods on `IO` class
* Keeping `ClockInterface()` method (and deprecating, maybe) as is and adding a new function `ClockIO` which returns an `m.IO` type
* I advocate **against** adding any other overloaded methods like `__setattr__` or `__setitem__`. Feels like the above is a more well-defined way of composing `IO`'s.

leonardt, 2020-02-26:
Re point 3, what if I want to add a port at at time (e.g. in a loop)?
Would it be something like this?
```
for i in range(N):
    io += m.IO(**{f"I{i}": m.In(m.Bit)})
```
Seems like 
```
for i in range(N):
    io[f"I{i}"] = m.In(m.Bit)
```
would be more convenient, but I guess there's an issue of mutation (overwriting an existing name whether that's an error or not.

We could use an explicit method like
```
for i in range(N):
    io.add(f"I{i}",  m.In(m.Bit))
```
but that still has the error on overwrite issue

rsetaluri, 2020-03-05:
We now have capability to add to an io like `io = m.IO(...) + m.IO(...)` or like `io += m.IO(...)`. Currently all `IO` objects are immutable (e.g. `+=` returns a *new* `IO` object), which is nice because it makes it easier to reason about them. However, it disallows certain things like:

```python
io = m.IO(x=..., y=...)
m.wire(io.x, io.y)
io += m.IO(...)  # previous wiring statements nullified
```

The wiring statements get wiped out because `+=` returns a new `IO` with new ref's, and the things we wired are no longer part of the `io`. We could support the above, but there's all sorts of funky stuff people can do like:

```
outer_io = m.IO(x=..., y=..)
m.wire(outer_io.x, outer_io.y)

class _Foo(m.Circuit):
    io = m.IO(z=..., w=...) + outer_io
     ...
```

I think it's hard to error/warn in all such cases, so for now it seems easier if it's just clear that `IO`'s are immutable and replacing an `IO` means that you completely nullify and previous wiring statements.

leonardt, 2020-03-05:
This issue comes up with the register file primitive and staging.

If we incrementally build a circuit, we can add a new port, then add new wiring, but this won't work with the IO being immutable and nullifying all old wiring.

The current alternative approach is to only manipulate the IO during first stages, then do all the wiring during the final stage once the IO is "finalized", which seems to work well enough.

leonardt, 2020-03-09:
One potential use case of appending ports, although it's not clear to me what the best path is:

I'm looking to add support for hierarchical instance references for the verification/debug support.  Basically, in verilog, you can use hierarchical instance syntax, e.g. `inst1.internal_inst2.O` to reference internal ports.  There's a few ways to support this pattern in magma:
1) Add the ability to do hierarchical selects in coreir (this would then just translate down into the equivalent verilog in the backend)
2) Add the ability to debug wire lifting in magma. When you do a hierarchical instance reference, it will edit the circuit to lift the desired port into the interface and then wire up the new instance port (so coreir wouldn't need to add support for this).

If we went with option 2, we would either need the ability to append ports to a circuit interface, or be able to easily copy circuit definitions.  Appending the interface has issues though, all other existing instances would need to tie off this new debug port (term).  Copying the circuit would probably be ideal, but possibly expensive? (Maybe okay since it's just a debug feature), but basically you would copy the IO, add the port, but then you would need to copy the entire wiring structure (I guess this wouldn't be so bad by just iterating over the interface and instances like we do in the backend, it's basically a "magma backend" for a circuit).

Any thoughts on whether we should go for 1 or 2 in terms of the higher level design? Basically should coreir add support for hierarchical instance references? CC @rdaly525

rsetaluri, 2020-03-24:
We also want to add an `__iter__` method to `IO`. This would be a convenience to users.

leonardt, 2020-07-24:
I think we've added the general features discussed here, we can open up follow-up issues for anything specific.

closed 2020-07-24 by leonardt