phanrahan/magma #598: Merge Aetherling's and Facebook's Reg/Mux/etc for Any Type

David-Durst, 2020-02-25:
Aetherling has a number of extensions to standard Magma/Mantle modules that handle any Magma type:
1. [LUTs](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/lut_any_type.py)
2. [MUXs](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/mux_any_type.py)
3. [Registers](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/register_any_type.py)
4. [ROMs](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/rom_any_type.py)
5. [Terms](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/term_any_type.py)
6. [SIPOs](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/sipo_any_type.py) 

They use the [hydrate/dehydrate modules](https://github.com/David-Durst/aetherling/blob/master/aetherling/modules/hydrate.py) for flattening types. Those call out to CoreIR.

Can we merge them with Facebook's implementation? Alternatively, can we replace them with Facebook's if Facebook's produces higher quality Verilog?

leonardt, 2020-04-21:
Let's use this issue for tracking the generator support for AnyTypes

leonardt, 2020-04-21:
We should provide the above listed generators as magma primitives, similar to how we implement the new Mux primitive here: https://github.com/phanrahan/magma/blob/master/magma/mux.py#L35-L64

leonardt, 2020-04-21:
They should use from_bits (unflatten) and as_bits (flatten) to implement the dehydrate/hydrate semantics (convert from any type to a bit vector and from a bit vector to any type).  These methods are now part of the Type interface, so we can expect any magma type to provide them.

leonardt, 2020-04-21:
The goal is to develop a `magma.primitives` submodule that provides these `Generator(T)s` using the new Generator2 pattern.  They will map to CoreIR primitives and provide Python simulation functions for the Python simulator.  The tests should ensure that the Python and CoreIR (or verilog) simulation matches, we can also regress on the generated coreir/verilog code.

CC @rsetaluri @rdaly525 @cdonovick if you have any input on this plan.  This is going to kick start the effort to migrate the core language primitives from mantle into magma (based on CoreIR), update them to the new Generator2 pattern (deprecating the Define<> pattern), and change their interfaces to be parametrized by type `T` (instead of bit width).

cdonovick, 2020-04-21:
My only real plan is to have an register primitive generated with `Register[T]` I am not entirely sure what the best `data_in` / `data_out` / `enable` interface is. 
```Python
reg = Register[T]()
x = T()
# This?
data_out = reg(data_in=x, enable=Bit(True))
# implicitly set enable?
data_out = reg(data_in=x)
# separate methods / properties?
data_out = reg.data_out
reg.data_in = x
reg.enable = Bit(True)
```

leonardt, 2020-04-23:
@David-Durst at some point, can you try out the new register primitive and see if it works for you? It would help to make sure it fits the requirements before we start work on the rest of the primitives.

David-Durst, 2020-04-24:
@leonardt Thank you! I will look at it soon.

leonardt, 2020-06-25:
Done! Feel free to open up any issues as things arise, the initial implementations of these are available now (see PRs for some info, will update docs soon)

closed 2020-06-25 by leonardt