phanrahan/magma #608: Doing operations on Clock types isn't supported by CoreIR

rsetaluri, 2020-03-02:
```python
import magma as m

class _Foo(m.Circuit):
    io = m.IO(I=m.In(m.Bit), O=m.Out(m.Bit), CLK=m.In(m.Clock))

class _Top(m.Circuit):
    io = m.IO(I=m.In(m.Bit), O=m.Out(m.Bit), CLK=m.In(m.Clock))
    foo = _Foo()
    foo.CLK <= m.clock(m.bit(io.CLK) & io.I)
    foo.I <= io.I
    io.O <= foo.O

m.compile("Top", _Top, output="coreir-verilog")
```

causes the error:

```
ERROR: _Top: Cannot wire together
  magma_Bit_and_inst0.out : Bit
  _Foo_inst0.CLK : coreir.clkIn
```

It looks like currently the `m.clock()` conversion function doesn't insert any logic that the compiler can use. One thing we could do is explicitly instance a conversion module.

rdaly525, 2020-03-02:
Yeah, currently you need to instance a coreir.strip module to 'cast' the clock type to a bit type. That being said, I am thinking it might be a good idea to remove the clock type in CoreIR.

rsetaluri, 2020-03-02:
@rdaly525 what's your recommendation for a quick, *temporary* solution?

rdaly525, 2020-03-02:
whenever you cast a clock to a bit instance the generator "coreir.strip" where you pass in the coreir type "coreir.clk" as strip's generator parameter (named 'type'). Ill see if I can write something simple using pycoreir to make sure all the APIs exist to do this.

rdaly525, 2020-03-02:
```
strip_gen = context.get_namespace("coreir").generators["strip"]
clk_type = context.named_types[("coreir","clk")]
strip_mod = strip_gen(type=clk_type)
```

The generated verilog should just look like a nop/passthrough

rsetaluri, 2020-03-05:
Seems this is fixed by #614

closed 2020-03-05 by rsetaluri