phanrahan/magma #609: test_tuple failure

phanrahan, 2020-03-02:
When I run the tests on my machine I get the following two failures in `test_tuple.py`.
```
=================================== FAILURES ===================================
_________________________________ test_nested __________________________________

    def test_nested():
        # Test for https://github.com/phanrahan/magma/issues/445
        def hierIO():
            class dictIO(Product):
                baseIO = make_baseIO()
                ctr = m.In(m.Bit)
            return dictIO
    
        def DefineCtrModule():
            class ctrModule(m.Circuit):
                name = "ctr_module"
                io = m.IO("ctr", m.In(m.Bit))
            return ctrModule
    
        def make_baseIO():
            class dictIO(Product):
                in0 = m.In(m.Bit),
                out0 = m.Out(m.Bit)
            return dictIO
    
        def DefineBaseModule():
            class baseModule(m.Circuit):
                name = "base_module"
                io = m.IO("baseIO", make_baseIO())
            return baseModule
    
        def DefineHier():
            class HierModule(m.Circuit):
                name = "hier_module"
                io = m.IO(hier=hierIO())
                @classmethod
                def definition(io):
                    baseM = DefineBaseModule()()
                    ctrM = DefineCtrModule()()
                    m.wire(baseM.baseIO, io.hier.baseIO)
                    m.wire(ctrM.ctr, io.hier.ctr)
            return HierModule
    
        baseMH = DefineHier()
>       m.compile("build/baseMH", baseMH, output="coreir-verilog")

tests/test_type/test_tuple.py:231: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
magma/compile.py:52: in compile
    compiler.compile()
magma/backend/coreir_compiler.py:39: in compile
    self.__compile_verilog()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <magma.backend.coreir_compiler.CoreIRCompiler object at 0xb2389de90>

    def __compile_verilog(self):
        # TODO(rsetaluri): Expose compilation to verilog in pycoreir rather than
        # calling the binary from the command line.
        cmd = _make_verilog_cmd(self.deps, self.basename, self.opts)
        ret = subprocess.run(cmd, shell=True).returncode
        if ret:
>           raise RuntimeError(f"CoreIR cmd '{cmd}' failed with code {ret}")
E           RuntimeError: CoreIR cmd 'coreir -l global -i /Users/hanrahan/git/magma/tests/test_type/build/baseMH.json -o /Users/hanrahan/git/magma/tests/test_type/build/baseMH.v' failed with code 1

magma/backend/coreir_compiler.py:56: RuntimeError
----------------------------- Captured stderr call -----------------------------
WARNING:magma:Supplying method 'definition' with new inline definition syntax is not supported, ignoring 'definition'
ERROR: Module: global.hier_module
  Type: {'hier':{'baseIO':{'out0':Bit}, 'ctr':BitIn}}
  Def? No has no def!

0   libcoreir.dylib                     0x000000010120060c _ZN6CoreIR7Context6setTopEPNS_6ModuleE + 124
1   libcoreir.dylib                     0x00000001012bc70c _ZN6CoreIR12loadFromFileEPNS_7ContextENSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEPPNS_6ModuleE + 29644
2   coreir                              0x0000000100ec58e1 main + 9969
3   libdyld.dylib                       0x00007fff6eb5f7fd start + 1
4   ???                                 0x0000000000000007 0x0 + 7
------------------------------ Captured log call -------------------------------
WARNING  magma:logging.py:87 Supplying method 'definition' with new inline definition syntax is not supported, ignoring 'definition'
________________________ test_tuple_nested_tuple_value _________________________

    def test_tuple_nested_tuple_value():
        def IFC0(params):
            return m.Product.from_fields("IFC0", {
                "port0": m.In(m.Bits[params['param0']]),
                "port1": m.In(m.Bits[params['param0']]),
                "port2": m.In(m.Array[params['param0'], m.Bits[2]]),
                "port3": m.In(m.Bits[params['param0']]),
                "port4": m.In(m.Bit),
                "port5": m.In(m.Bit),
                "port7": m.In(m.Bit),
                "port8": m.In(m.Bit),
                "port9": m.In(m.Bit),
                "port10": m.In(m.Bits[m.bitutils.clog2(params['param0'])]),
            })
    
        def IFC1(params):
            dictOut = {"port4": m.Out(m.Bit)}
            return m.Product.from_fields("IFC1", dictOut)
    
        def DefineMyCircuit(params):
            class MyCircuit(m.Circuit):
                io = m.IO(IFC0=IFC0(params).flip())
            return MyCircuit
    
        def DefineTop(params):
            class Top(m.Circuit):
                io = m.IO(IFC1=IFC1(params))
                @classmethod
                def definition(io):
                    m.wire(io.IFC1.port4, DefineMyCircuit(params)().IFC0.port4)
            return Top
    
>       m.compile("top", DefineTop({'param0': 5}))

tests/test_type/test_tuple.py:266: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
magma/compile.py:52: in compile
    compiler.compile()
magma/backend/coreir_compiler.py:39: in compile
    self.__compile_verilog()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <magma.backend.coreir_compiler.CoreIRCompiler object at 0xb1dc14f90>

    def __compile_verilog(self):
        # TODO(rsetaluri): Expose compilation to verilog in pycoreir rather than
        # calling the binary from the command line.
        cmd = _make_verilog_cmd(self.deps, self.basename, self.opts)
        ret = subprocess.run(cmd, shell=True).returncode
        if ret:
>           raise RuntimeError(f"CoreIR cmd '{cmd}' failed with code {ret}")
E           RuntimeError: CoreIR cmd 'coreir -l global -i /Users/hanrahan/git/magma/tests/test_type/top.json -o /Users/hanrahan/git/magma/tests/test_type/top.v' failed with code 1

magma/backend/coreir_compiler.py:56: RuntimeError
----------------------------- Captured stderr call -----------------------------
WARNING:magma:Supplying method 'definition' with new inline definition syntax is not supported, ignoring 'definition'
ERROR: Module: global.Top
  Type: {'IFC1':{'port4':Bit}}
  Def? No has no def!

0   libcoreir.dylib                     0x00000001025f460c _ZN6CoreIR7Context6setTopEPNS_6ModuleE + 124
1   libcoreir.dylib                     0x00000001026b070c _ZN6CoreIR12loadFromFileEPNS_7ContextENSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEPPNS_6ModuleE + 29644
2   coreir                              0x00000001022b88e1 main + 9969
3   libdyld.dylib                       0x00007fff6eb5f7fd start + 1
4   ???                                 0x0000000000000007 0x0 + 7
------------------------------ Captured log call -------------------------------
WARNING  magma:logging.py:87 Supplying method 'definition' with new inline definition syntax is not supported, ignoring 'definition'
```

rsetaluri, 2020-03-02:
Looks like the conversion to the new style syntax wasn't done correctly. Not sure why it's passing on travis...Will f ix.

rsetaluri, 2020-03-02:
@phanrahan can you verify that this is no longer present if you pull from master? (See #610.)

phanrahan, 2020-03-02:
Works now. Thanks!

closed 2020-03-02 by rsetaluri