phanrahan/magma #631: pytest fails due to coroutine

phanrahan, 2020-03-11:
```
% pytest
/Users/hanrahan/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
ImportError while loading conftest '/Users/hanrahan/git/magma/conftest.py'.
conftest.py:2: in <module>
    from magma import clear_cachedFunctions
magma/__init__.py:45: in <module>
    from .circuit import *
magma/circuit.py:28: in <module>
    from magma.syntax.combinational import combinational
magma/syntax/__init__.py:3: in <module>
    from .coroutine import coroutine
magma/syntax/coroutine.py:5: in <module>
    from staticfg import CFGBuilder
E   ModuleNotFoundError: No module named 'staticfg'
```

leonardt, 2020-03-11:
`staticfg` was added to the `setup.py` as a requirement.  If you haven't rerun setup.py since pulling you can do `pip install staticfg`.  However to run the coroutine test, you'll need my fork of staticfg by doing `pip install -r requirements.txt` (inside this repo)

phanrahan, 2020-03-18:
That worked.

closed 2020-03-18 by phanrahan