phanrahan/magma #697: @= wiring bug with protocol

leonardt, 2020-04-28:
```
    class FooMeta(m.MagmaProtocolMeta):
        def _to_magma_(cls):
            return cls.T

        def _qualify_magma_(cls, direction: m.Direction):
            return cls[cls.T.qualify(direction)]

        def _flip_magma_(cls):
            return cls[cls.T.flip()]

        def _from_magma_value_(cls, val: m.Type):
            return cls(val)

        def __getitem__(cls, T):
            return type(cls)(f"Foo{T}", (cls, ), {"T": T})

    class Foo(m.MagmaProtocol, metaclass=FooMeta):
        def __init__(self, val: Optional[m.Type] = None):
            if val is None:
                val = self.T()
            self._val = val

        def _get_magma_value_(self):
            return self._val

        def non_standard_operation(self):
            v0 = self._val << 2
            v1 = m.bits(self._val[0], len(self.T)) << 1
            return Foo(v0 | v1 | m.bits(self._val[0], len(self.T)))

    class Bar(m.Circuit):
        io = m.IO(foo=m.In(Foo[m.Bits[8]]), O=m.Out(m.Bits[8]))
        io.O @= io.foo.non_standard_operation()
```

produces

```
ERROR:magma:Cannot wire <tests.test_magma_protocol.test_foo_type_magma_protocol.<locals>.Foo object at 0x1172619b0> (type=<class 'tests.test_magma_protocol.test_foo_type_magma_protocol.<locals>.Foo'>) to .O (type=In(Bits[8])) because <tests.test_magma_protocol.test_foo_type_magma_protocol.<locals>.Foo object at 0x1172619b0> is not an Array
ERROR:magma:Output port Bar.O not driven
```

I suspect the `Array.wire` logic, which is invoked directly by the imatmul operator skips the type dispatch on protocols.  One option is to add a decorator for wire methods that unpacks protocol types.

leonardt, 2020-04-28:
For now, a workaround exists with `m.wire(io.foo.non_standard_operation(), io.O)`

rsetaluri, 2020-04-28:
Yeah this is a tough one because it requires that check in each `imatmul` function. In general, there are some discrepancies between the behavior of `m.wire(x, y)`, `x @= y`, and `x.wire(y)` (though the last 2 are probably almost always the same). We should try to clean up/unify this. Caleb's idea's around having free functions do all this work for us might be best, i.e. `magma.value(x) == x is protocol ? x._magma_value_() : x`