phanrahan/magma #704: Python 3.8 issues

splhack, 2020-05-27:
`pytest tests` has two issues with Python 3.8.2 (Ubuntu 20.04 default)

First. test_old_io_syntax metadata lineno differences

    diff --git a/tests/test_deprecated/test_old_io_syntax/gold/test_for_loop_def.json b/tests/test_deprecated/test_old_io_syntax/gold/test_for_loop_def.json
    index 1b8d5f27..58cabd81 100644
    --- a/tests/test_deprecated/test_old_io_syntax/gold/test_for_loop_def.json
    +++ b/tests/test_deprecated/test_old_io_syntax/gold/test_for_loop_def.json
    @@ -8,7 +8,7 @@
               ["I1","BitIn"],
               ["O","Bit"]
             ]],
    -        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"55"}
    +        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"54"}
           },
           "main":{
             "type":["Record",[
    diff --git a/tests/test_deprecated/test_old_io_syntax/gold/test_interleaved_instance_wiring.json b/tests/test_deprecated/test_old_io_syntax/gold/test_interleaved_instance_wiring.json
    index a0978e76..21b8d494 100644
    --- a/tests/test_deprecated/test_old_io_syntax/gold/test_interleaved_instance_wiring.json
    +++ b/tests/test_deprecated/test_old_io_syntax/gold/test_interleaved_instance_wiring.json
    @@ -8,7 +8,7 @@
               ["I1","BitIn"],
               ["O","Bit"]
             ]],
    -        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"87"}
    +        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"86"}
           },
           "main":{
             "type":["Record",[
    diff --git a/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def.json b/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def.json
    index 157f94a0..a70e1d30 100644
    --- a/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def.json
    +++ b/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def.json
    @@ -8,7 +8,7 @@
               ["I1","BitIn"],
               ["O","Bit"]
             ]],
    -        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"14"}
    +        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"13"}
           },
           "main":{
             "type":["Record",[
    diff --git a/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def_class.json b/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def_class.json
    index 76807617..8d5bf14c 100644
    --- a/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def_class.json
    +++ b/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def_class.json
    @@ -8,7 +8,7 @@
               ["I1","BitIn"],
               ["O","Bit"]
             ]],
    -        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"14"}
    +        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"13"}
           },
           "Main":{
             "type":["Record",[

Second. test_jtag.py recursion error. regardless with or without #703 

    pytest tests/test_syntax/test_coroutine/test_jtag.py
    E               RecursionError: maximum recursion depth exceeded while calling a Python object

leonardt, 2020-05-27:
Thanks for reporting this, I'll take a look.  It's likely that the metadata lineno thing can be updated, but it might be a bit annoying to make sure the tests work with 3.7 and 3.8, I"ll have to look at the code to see why this is happening, but it seems like Python may have changed how it stores this metadata in the AST.

Not sure about the jtag test, will take a look

cdonovick, 2020-06-13:
@leonardt one more reason to move LibCST

splhack, 2020-07-29:
No longer see the issues on 6adc6ee7fb3989e8f7f46983645a7f8ff1e89cc6

closed 2020-07-29 by splhack