phanrahan/magma #731: sequential2: Register access shortcut?

splhack, 2020-06-22:
sequential can use register like this (`self.count` instead of `self.count.I` and `self.count.O`)

    @m.circuit.sequential(async_reset=False)
    class Test1:
        def __init__(self):
            self.count: m.SInt[16] = m.sint(0, 16)
        def __call__(self) -> m.SInt[16]:
            self.count = self.count + 1
            return self.count

sequential2 allows to write similar code, but `self.count.O + 1` instead of `self.count + 1`. can it be like sequential's? (or am I wrong about `__init__` Register declaration?)

    @m.sequential2()
    class Test2:
        def __init__(self):
            self.count = m.Register(T=m.SInt[16], init=m.sint(0, 16))()
        def __call__(self) -> m.SInt[16]:
            self.count = self.count.O + 1
            return self.count

leonardt, 2020-06-24:
This is a bug, working on a fix

leonardt, 2020-06-25:
Should be fixed by #737, thanks for testing this out, let us know if you run into more problems!

closed 2020-06-25 by leonardt

splhack, 2020-06-25:
ran into #739 😄