phanrahan/magma #735: pytest slow

rsetaluri, 2020-06-22:
Finding that running full pytest is slow. Just did a quick profile. Most of the tests are ~1s, except for the following:

```
test_log.prof | 4.451536
test_jtag.prof | 9.470764
test_array_of_bits[coreir].prof | 9.599056
test_array_of_bits[coreir-verilog].prof | 9.834311
test_sdram.prof | 49.562007
```

The remainder of the suite takes ~40s cumulatively; these 5 tests take ~80s.

Is there anything we can do to get these down? If they're fundamentally this long, that's fine, but we should mark them long tests or something, so that for local dev iteration we can skip them. Ideally, in general we could use https://docs.pytest.org/en/stable/example/markers.html to mark which tests are "core" or "regression" or "smoke." For example, I often just want to make sure that I didn't break any core feature.

@leonardt thoughts?

leonardt, 2020-06-24:
array of bits tests will likely be improved by porting to sequential2 and using the new register(T) primitive, so I would suggest porting that code over and deprecating the old functionality.  The log test may have to do with slowness with writing to disk, not sure what we could improve there, maybe using a named pipe? But we'd have to pass that through to the verilog TB some how.  Not sure about the SDRAM code gen off the top of my head, we'll have to profile that one.

leonardt, 2020-06-26:
We've addressed the issues for the major sources of problems, test_log hasn't been sped up but looking into it, most of the time is spent inside waiting for the verilator test benches to run.  I'm guessing they're slow because of IO (printing to stdout or to a file), and I'm not sure if we can really improve this without making the tests less useful. Should we close this for now? 4s isn't so bad compared to the others I would say.

leonardt, 2020-08-12:
Going to close this for now, but feel free to reopen if it's still an issue.

closed 2020-08-12 by leonardt