phanrahan/magma #752: ND arrays + indexing (feature request)

kavyasreedhar, 2020-06-30:
I'm able to create a ND array using m.join, but I wasn't able to index into the array like arr[0][0][0] and have a separate indexing function that flattens the address. I was wondering if support could be added for multidimensional array indexing or if it already exists and I'm just missing it (and also the same with hwtypes support for a multidimensional BitVector)? Thanks!

leonardt, 2020-06-30:
Hmm, not sure about the m.join indexing issue, can you post a little code snippet that demonstrates how you're constructing the nd array?

We don't have a built-in ndarray type ala numpy, but that's been requested before (I'll try to find the old issue) and is something we definitely want to add.  Right now the only way to emulate an n-d array is to create the equivalent nested array type, (e.g. `m.Array[5, m.Array[6, m.Bits[8]]]` will create a 5x6 array of bytes).  You can index a value of that type as expected, e.g. `arr[3][5] + 1`.

CC @cdonovick about adding ndarray (multidimensional bit vector) support for hwtypes, is something you'd be interested in?

leonardt, 2020-06-30:
Here's the old feature request (from @cdonovick actually!) https://github.com/phanrahan/magma/issues/165, we will bump this up in priority

kavyasreedhar, 2020-06-30:
Awesome, thanks! I think what you wrote above is enough for my use case, so I will go forward with that.

rsetaluri, 2020-07-01:
Citing @cdonovick from #165 

> Feature Request: ndarray
> Multi dimensional arrays declared with shape tuple similar to numpy would be nice

rsetaluri, 2020-07-01:
To understand, is the ask for just a sugar-y constructor for ND arrays of type T (with no new underlying features), ala

```python
def NDArray(T, *widths):
    for width in widths:
        T = Array[width, T]
    return T
```
(ignoring `()` vs `[]` syntax, figure it would be easy enough to transform into that)

On the other hand, I could see a new `NDArray` type constructor entirely. It's not clear to me what benefits that would have. One thing I can think of off the top of my head is tuple-indexing directly. So we could write `x[0, 1, 2]` instead of `x[0][1][2]`. Unsure if this is useful, or what other benefits a separate type would have...

leonardt, 2020-07-01:
convenient constructor and tuple indexing seems like two reasonable features.  another might be element wise operations on the leaf data type? (ala numpy element-wise operators)

leonardt, 2020-07-22:
https://github.com/phanrahan/magma/pull/777 adds a prototype for this.  

Here's the small working example:
```python
    class Main(m.Circuit):
         io = m.IO(I=m.In(m.NDArray[3, 5]), O=m.Out(m.NDArray[5, 3]))
         for i in range(3):
             for j in range(5):
                 io.O[j, i] @= io.I[i, j]
```

@kavyasreedhar how does this look? Do you need more than just tuple based construction and indexing? Right now it assumes the leaf type is Bit, we could update this to accept a generic type, but perhaps that's not necessary for your use case?

cdonovick, 2020-07-22:
I would like add array to hwtypes.  I am pretty low bandwidth right now though.  

My Ideal syntax would basically follow numpy:

```python
# for brevity I am going to us in to mean isinstance
assert Array[N, T] is Array[(N,), T]
a  = Array[(N, M), T](..)
assert a[i] in Array[M, T] 
assert a[:, j] in Array[N, T]
assert a[i, j] in T
```

Additionally I would like to support slicing, which would make it easy to define "convolutional" architectures.  Basically braid except people other than Pat would understand it. 

Ideally we would also support `...` but its not a high priority.
In numpy if the number of indices are less than the number of dimensions the index is padded on the right with `:`.
A `...` means put the padding here. 
```Python
a = Array[(N, M, P), T](..)
assert a[i] == a[i, ...] == a[i, :, :] in Array[(M, P), T]
assert a[..., i, :] == a[:, i, :] in Array[(N, P), T]
assert a[..., i] == a[:, :, i] in Array[(N, M), T]
```

leonardt, 2020-07-22:
Using a tuple for what was the N index makes sense, that would make it easier to extend the current array type to support it, I’ll refactor the prototype to do that.

leonardt, 2020-07-23:
Ok here's the new example:
```python
    class Main(m.Circuit):
         io = m.IO(I=m.In(m.Array[(3, 5), m.Bit]),
                   O=m.Out(m.Array[(5, 3), m.Bit]))
         for i in range(3):
             for j in range(5):
                 io.O[j, i] @= io.I[i, j]
```

I'll probably hold off on the fancier indexing patterns unless they're needed immediately, but this adds tuple indexing and tuple for the N parameter.  I basically just extended the ArrayMeta getitem to handle the tuple key and Array getitem to handle tuple indices.  We should probably add some safety check, so when `isinstance(N, tuple)` during type construction, the tuple indexing is enabled, otherwise throw an error if a user tries to tuple index a normal array (right now they'll probably get some strange, confusing error if they try to index with a tuple)

leonardt, 2020-07-23:
Marking NDArrays as mentioned above (for error messages) will also be useful to the compiler as it can use it to generate an NDArray type in CoreIR (which we're planning to add for a more compact representation for improved performance, as well for code generated multi-dimensional verilog arrays for performance/readability)

kavyasreedhar, 2020-07-23:
I think slicing would also be really useful. 

I think Bit is okay for the type, but I vaguely remember that arithmetic/bitwise ops need to be done with Unsigned or Signed. If that is the case, support for Unsigned would be really helpful so we can do something like io.arr[i, j] = io.arr[i, j] + 1 (the syntax may be off in that, but hopefully the idea is clear).

leonardt, 2020-07-23:
The latest version accepts an arbitrary type as a parameter so using unsigned or signed should work.  I'll look into slicing, can you provide a code example of the kind of slicing you want to support? There's a lot of a general patterns but maybe we'll be able to start with a simpler one for your use case.

kavyasreedhar, 2020-07-23:
Cool, thanks!

For slicing, it would be nice if something like the following were supported (again, syntax may be off / is not too important, but hopefully the intended functionality comes through):

a is some m.Array[(4, 5, 3), m.Bit]
b is some m.Array[(5, 3), m.Bit]
c is some m.Array[(3, 5, 3), m.Bit]
d is some m.Array[(3, 2), m.Bit]
(I'm assuming 4 is the innermost dimension and 5 the outermost for the below examples)
I'm using inclusive lower bound and exclusive upper bound, but that particular syntax isn't too important either.
a[0:2] = c (slicing for a range in one dimension)
a[0:2][2:3][2] = d (slicing for a range in multiple dimensions, with one dimension just having 1 dimension)
Also, it'd be nice if d didn't have to specify it's dimensions as (3, 2, 1) for this slicing example.
Also, any mix and match of the above (so any supported ranges and range/single indexing combos for all the dimensions).

I'll update if I think of more / look at our code and see what else we use.

leonardt, 2020-07-23:
Having the first index be the inner most dimension matches numpy, my example had it reversed, so I updated the code to match your example/numpy, I think that's a reasonable choice (people should be familiar with that)

leonardt, 2020-07-23:
Hmm, those slicing examples may just work out of the box or with minimal work, I'll try them out.  It's simpler than using the `...` which might be able to capture some of those examples, but if you're okay with using non-tuple indexing for slicing (i.e. using separate `[]` for each dimension, then that should be simpler to implement)

kavyasreedhar, 2020-07-23:
Yeah, that sounds great, I think I'm not too worried about the syntax and more about the functionality, so separate [] for each dimension should be great to start. I also think another nice feature would be [:] to mean the entire dimension or instead support something like [0:len(self.variable)].

On that train of thought, I also think support for having non-integer index slices would be the most helpful (so something like [0:self.addr] where self.addr is an input) because usually I don't know the indices for slicing off the bat and am using a variable or an input to index.

leonardt, 2020-07-23:
Ok, so I think we actually need to use tuple indexing, that's how numpy handles this type of multi-dimensional indexing pattern.  The problem with using chained get items ([]) is that when you select a range, it returns an array with the outer dimension trimmed, and there's no way for us to know that the next getitem is supposed to be broadcast to the inner dimensions rather than the returned array.

Here's an example that illustrates the issue (how numpy handles it):

```python
>>> a = np.arange(60).reshape(5, 4, 3)
>>> a[1:5][0:2]
array([[[12, 13, 14],
        [15, 16, 17],
        [18, 19, 20],
        [21, 22, 23]],

       [[24, 25, 26],
        [27, 28, 29],
        [30, 31, 32],
        [33, 34, 35]]])
>>> a = np.arange(60).reshape(5, 4, 3)
>>> a
array([[[ 0,  1,  2],
        [ 3,  4,  5],
        [ 6,  7,  8],
        [ 9, 10, 11]],

       [[12, 13, 14],
        [15, 16, 17],
        [18, 19, 20],
        [21, 22, 23]],

       [[24, 25, 26],
        [27, 28, 29],
        [30, 31, 32],
        [33, 34, 35]],

       [[36, 37, 38],
        [39, 40, 41],
        [42, 43, 44],
        [45, 46, 47]],

       [[48, 49, 50],
        [51, 52, 53],
        [54, 55, 56],
        [57, 58, 59]]])
# 1:4 returns an array of dimension (3, 4, 3)
>>> a[1:4]
array([[[12, 13, 14],
        [15, 16, 17],
        [18, 19, 20],
        [21, 22, 23]],

       [[24, 25, 26],
        [27, 28, 29],
        [30, 31, 32],
        [33, 34, 35]],

       [[36, 37, 38],
        [39, 40, 41],
        [42, 43, 44],
        [45, 46, 47]]])
# Chaining a slice will slice the return the sliced (3, 4, 3) array of dimension (2, 4, 3)
>>> a[1:4][0:2]
array([[[12, 13, 14],
        [15, 16, 17],
        [18, 19, 20],
        [21, 22, 23]],

       [[24, 25, 26],
        [27, 28, 29],
        [30, 31, 32],
        [33, 34, 35]]])
```

If instead we wanted to have `0:2` broadcast to the middle dimension in the above example, we would use tuple indexing
```python
>>> a[1:4, 0:2]
array([[[12, 13, 14],
        [15, 16, 17]],

       [[24, 25, 26],
        [27, 28, 29]],

       [[36, 37, 38],
        [39, 40, 41]]])
```

now we get an output of dimension (3, 2, 3) which is what I believe you expect for your above example.  The reason this works is that when you use tuple indexing, we know that we can broadcast the slices to the inner dimensions.  Otherwise, when we return something like a (3, 4, 3) array and you use getitem again, we have to think that the slice is being used on the outermost dimension (because this is what happened the first time).

I don't think this should affect your use case besides just having to change the syntax to tuple style.

kavyasreedhar, 2020-07-23:
Oh no worries, I think the tuple style indexing sounds good and it is helpful to be consistent with numpy too, so that sounds great!

leonardt, 2020-07-23:
Ok, latest code in the PR adds the slice feature for your example, it seems to work but when you get a chance to try it out there may be issues so let me know if you find any problems.  One thing to note is that python use non-inclusive range (so the indexing pattern will be slightly different than verilog).

There's one minor issue, the test wiring logic works and the generated verilog looks good, but I see an error reported:
```
  logging.py                  87 ERROR    May not mutate array, trying to replace a1[(2, slice(2, 5, None), slice(0, 2, None))] (AnonymousValue_140275947723016) with AnonymousValue_140275947722680
  logging.py                  87 ERROR    May not mutate array, trying to replace a1[(2, slice(0, 2, None), slice(0, 2, None))] (AnonymousValue_140275947723408) with AnonymousValue_140275947722960
  logging.py                  87 ERROR    May not mutate array, trying to replace a1[(3, slice(None, None, None), slice(0, 2, None))] (AnonymousValue_140275947723800) with AnonymousValue_140275947722792
  logging.py                  87 ERROR    May not mutate array, trying to replace a1[(slice(0, 2, None), slice(None, None, None), slice(0, 2, None))] (AnonymousValue_140275947808136) with AnonymousValue_1402759
```
which looks like another manifestation of the issue described here https://github.com/phanrahan/magma/issues/547 (with the original fix here https://github.com/phanrahan/magma/pull/552).  It may be the way the slicing works with getitem is throwing off the "orig" check for the setitem operator (since we're using imatmul with slices and slice constructs new values to return), I'll see if I can figure out how to get the safety check to be slice aware.

leonardt, 2020-07-24:
Above error fixed by recursively checking contents (slices return new values that wrapped the sliced children, so we can't do a simple is check)

kavyasreedhar, 2020-07-25:
Awesome, thanks! I just wanted to confirm, is this expected to work with the index being a input variable as well?

leonardt, 2020-07-27:
I haven't added that feature yet.  I think we can support setting a single index with variable , but for slicing we'll need to use get_slice/set_slice for the same reasons why we needed to do it for regular arrays.

leonardt, 2020-08-07:
@kavyasreedhar the ndarray feature has been merged into master, let me know how it works out for you and if you have any feedback for improvements.  Let's use new issues to track anything that comes up.

closed 2020-08-07 by leonardt

kavyasreedhar, 2020-08-07:
Sounds good, thanks Lenny!