phanrahan/magma #753: Setting a slice of a bit vector (feature request)

kavyasreedhar, 2020-06-30:
@leonardt @cdonovick 

Is it possible to add setting a part of a register instead of having to specify the whole signal? I was talking with Caleb about a NotImplementedError I was getting for set_item today. It would be nice if I could do something like

self.mem[x : x + 16] = data_in 

(where self.mem is a BitVector) instead of having to do 

self.mem = concat(self.mem[ : x], data_in, self.mem[ x + 16 : ]) 

because as Caleb pointed out, I would also have to watch out for the case when I have a 0 length slice which would cause errors, so this code would actually need to be converted into an if statement where I check if (in this example) x is 0 or the max address and separate out my code into 3 separate cases with different concat scenarios. Caleb suggested asking for the feature on "SequentialRegister", if that's possible? Thanks!

leonardt, 2020-06-30:
This should be possible, but to clarify, is x here a python constant value? Or is it a dynamic index?

kavyasreedhar, 2020-06-30:
x is an input to the module (like addr). The actual code I had if that's helpful was  self.mem[addr * data_width : (addr + 1) * data_width] = data_in, where addr is an input, data_width is a parameter, and data_in is also an input.

cdonovick, 2020-06-30:
@leonardt I think we could special case dynamic indices for `getitem :: BV[n] -> BV[k] -> Bit where n == 1 << k`  Its a total function that can be defined with shifts so it shouldn't be a problem.   Similar for setitem.

cdonovick, 2020-06-30:
Obviously magma could something smarter if it wants to

cdonovick, 2020-06-30:
This wouldn't address how this precise scenario where a slice being extracted but I think we could something similar.

leonardt, 2020-06-30:
Right now magma provides an `m.slice` operator for these dynamic offset, constant width slices, see the documentation here https://github.com/phanrahan/magma/blob/c0a50856f93c2732e392f87920cb0eff0e99ea70/magma/primitives/slice.py#L5, I think this would work for this use case?

This case is generally much simpler than the general getitem/setitem slice support.  For example, in this case, we need to inspect the slice keys (start, stop) and observe that they're magma values.  Then we need to analyze them (trace the expression graph?) to determine it's a constant width slice expression.  Then we could generate the code that the m.slice operator generates.  However, in general, the width of the slice might be non-constant (e.g. instead of addr + 1, we could have adds + x where x is a dynamic input to the circuit).  I'm not sure if we can support this general case given magma's current type system, because this would require encoding a dynamically sized output (the width of the output bit vector depends on the dynamic value of x).

Given this constraint, I would advocate the use of a special constant width slice operator like `slice` that allows us to handle the case of interest here.  Alternatively we can add support for this specific case in the getitem/setitem logic, but we'll need to implement the constant width tracing analysis described above (to determine the width of the output).

One limitation with the current slice operator is it only operates on output, so we would need a strategy to allow it to work with inputs (for the setitem pattern)

cdonovick, 2020-06-30:
I like the idea of a special slice object.  Let me think about how best to pull this back into hwtypes.

leonardt, 2020-07-01:
Here's the proposed slice operator interface
```
    class TestSequential2Slice:
         def __init__(self):
             self.mem = m.Register(T=m.Bits[8 * 8])()

         def __call__(self, write_addr: m.UInt[3], write_data: m.Bits[8],
                      read_addr: m.UInt[3]) -> m.Bits[8]:
             read_data = m.get_slice(self.mem, m.zext(read_addr, 3) * 8, 8)
             self.mem = m.set_slice(self.mem, write_data,
                                    m.zext(write_addr, 3) * 8, 8)
             return read_data
```

fortunately, this works out of the box with sequential2

leonardt, 2020-07-01:
set_slice operator has been merged into master, @kavyasreedhar can you try it out and see if it works for you? if you're installed via pip, I can make a new magma release, just let me know.

kavyasreedhar, 2020-07-01:
Thanks! Would you be able to make a new magma release?

kavyasreedhar, 2020-07-01:
I cloned the magma repo and installed in my environment. I think set_slice is okay because I'm not getting an error there when running, but I am having issues testing the magma with a get_slice helper function from Caleb. 

Could a similar operator be added to get a part of a register, e.g. data_out = get_slice(self.mem, addr * data_width, data_width) which is equivalent to something like data_out = self.mem[addr * data_width : (addr + 1) * data_width] where self.mem is a bit vector, addr is an input bit vector, and data_width is a parameter python constant?

leonardt, 2020-07-01:
get_slice should do what you're asking, can you post the error message you're getting?

There is one difference from your code to my example, in mine we extend the address to accommodate the extra bits needed for the multiplication, perhaps that's the problem? (Not sure what the width of your addr value is)

leonardt, 2020-07-01:
magma-lang 2.0.62 released with get/set slice operators

kavyasreedhar, 2020-07-01:
The error I am seeing is here:

/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
Traceback (most recent call last):
  File "/nobackup/skavya/magma/magma/ast_utils.py", line 54, in compile_function_to_file
    exec(code, defn_env)
  File ".magma/SRAMStub_comb.py", line 1, in <module>
    class SRAMStub_comb(m.Circuit):
  File ".magma/SRAMStub_comb.py", line 7, in SRAMStub_comb
    data_out_0 = get_slice(self_mem_I_0, io.addr * data_width, data_width)
  File "sram_stub.py", line 10, in get_slice
    return (data >> addr)[:width]
  File "/nobackup/skavya/magma/magma/bits.py", line 408, in __rshift__
    raise e from None
  File "/nobackup/skavya/magma/magma/bits.py", line 406, in __rshift__
    return self.bvlshr(other)
  File "/nobackup/skavya/magma/magma/bits.py", line 42, in wrapped
    other = _coerce(type(self), other)
  File "/nobackup/skavya/magma/magma/bits.py", line 33, in _coerce
    raise InconsistentSizeError('Inconsistent size')
hwtypes.bit_vector_abc.InconsistentSizeError: Inconsistent size

Traceback (most recent call last):
  File "/nobackup/skavya/magma/magma/ast_utils.py", line 54, in compile_function_to_file
    exec(code, defn_env)
  File ".magma/SRAMStub_comb.py", line 1, in <module>
    class SRAMStub_comb(m.Circuit):
  File ".magma/SRAMStub_comb.py", line 7, in SRAMStub_comb
    data_out_0 = get_slice(self_mem_I_0, io.addr * data_width, data_width)
  File "sram_stub.py", line 10, in get_slice
    return (data >> addr)[:width]
  File "/nobackup/skavya/magma/magma/bits.py", line 408, in __rshift__
    raise e from None
  File "/nobackup/skavya/magma/magma/bits.py", line 406, in __rshift__
    return self.bvlshr(other)
  File "/nobackup/skavya/magma/magma/bits.py", line 42, in wrapped
    other = _coerce(type(self), other)
  File "/nobackup/skavya/magma/magma/bits.py", line 33, in _coerce
    raise InconsistentSizeError('Inconsistent size')
hwtypes.bit_vector_abc.InconsistentSizeError: Inconsistent size

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "sram_stub.py", line 69, in <module>
    tester = fault.Tester(sram_magma(family=family.MagmaFamily())())
  File "/nobackup/skavya/python_envs/hwtypes/src/peak/peak/features.py", line 110, in __call__
    res = self.fc(family, *args, **kwargs)
  File "sram_stub.py", line 34, in modules_fc
    @family.assemble(locals(), globals())
  File "/nobackup/skavya/python_envs/hwtypes/src/peak/peak/family.py", line 223, in deco
    cls = m.circuit.sequential(cls, env=env)
  File "/nobackup/skavya/magma/magma/syntax/sequential.py", line 552, in sequential
    return wrapped_sequential(m.AsyncReset, cls, combinational)
  File "/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/ast_tools/stack.py", line 116, in wrapped_0
    return fn(env, *args, **kwargs)
  File "/nobackup/skavya/magma/magma/syntax/sequential.py", line 501, in _sequential
    circuit_def = circuit_def_constructor(combinational_decorator)
  File ".magma/make_SRAMStub.py", line 8, in make_SRAMStub
    class SRAMStub(m0.Circuit):
  File ".magma/make_SRAMStub.py", line 16, in SRAMStub
    self_mem_O: BitVector[256]) ->(BitVector[256], WideData):
  File "/nobackup/skavya/magma/magma/syntax/combinational.py", line 272, in wrapped
    return wrapped_combinational(fn)
  File "/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/ast_tools/stack.py", line 116, in wrapped_0
    return fn(env, *args, **kwargs)
  File "/nobackup/skavya/magma/magma/syntax/combinational.py", line 233, in _combinational
    defn_env)
  File "/nobackup/skavya/magma/magma/ast_utils.py", line 59, in compile_function_to_file
    raise Exception(f"Error occured when executing m.circuit.combinational function {func_name}, see above")
Exception: Error occured when executing m.circuit.combinational function SRAMStub_comb, see above

kavyasreedhar, 2020-07-01:
The width of my addr is log2(depth), where depth is another constant parameter. The test module (it's fairly short) if it's helpful is here: https://github.com/StanfordAHA/lake/blob/accessor/lake/modules/hwtypes/sram_stub.py

leonardt, 2020-07-01:
Oh, I'm not sure about the get_slice helper function you are using, can you try the built-in `m.get_slice`?

leonardt, 2020-07-01:
But the above error looks like a type error related to the size of the shift arguments (data and addr are inconsistent sizes)

kavyasreedhar, 2020-07-01:
The get_slice helper function was from Caleb. I'm trying to use hwtypes so I can get the functional model and the magma/verilog from the same code, so I think that helper function should correspond to the m.get_slice function like set_slice and m.set_slice. My data and addr are different sizes because my data_width is 64 bits, but my depth is 4 so I only need 2 bits to represent all possible addresses. Is that causing a problem?

kavyasreedhar, 2020-07-01:
Oh actually, I think extending get_slice like we had to do for set_slice got past that issue! I added addr = addr_.zext(data.size - addr_.size) in get_slice and it seems okay. I run into another problem, but I think that's unrelated, so I think those tentatively are working for me now!

kavyasreedhar, 2020-07-01:
Not sure if this is the best place to post because I am not sure if it's related to get/set_slice, but I am having a problem with these comparisons now: if (cen == Bit(1)) and (wen == Bit(1)):

Is there different syntax for the comparison or should I not be comparing to the Bit hwtype?

Traceback (most recent call last):
  File "/nobackup/skavya/magma/magma/ast_utils.py", line 54, in compile_function_to_file
    exec(code, defn_env)
  File ".magma/SRAMStub_comb.py", line 1, in <module>
    class SRAMStub_comb(m.Circuit):
  File ".magma/SRAMStub_comb.py", line 9, in SRAMStub_comb
    data_out_2 = phi([data_out_1, data_out_0], io.cen == Bit(1) and io.wen ==
  File "/nobackup/skavya/magma/magma/bit.py", line 154, in __bool__
    raise NotImplementedError("Converting magma bit to bool not supported")
NotImplementedError: Converting magma bit to bool not supported

leonardt, 2020-07-01:
You should use & instead of `and`.  `and` is sort of similar to Verilog's && operator (logical and), except in Python we don't support this pattern because we can't override the operator.  So instead you should do a bit-wise & of the two equality expressions (it will be logically equivalent)

kavyasreedhar, 2020-07-01:
Oh I did try & instead of and, but that also resulted in a (different) error that I wasn't sure about:

Traceback (most recent call last):
  File "/nobackup/skavya/magma/magma/circuit.py", line 538, in __init__
    context = _definition_context_stack.peek()
  File "/nobackup/skavya/magma/magma/common.py", line 19, in peek
    return self._stack[-1]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "sram_stub.py", line 76, in <module>
    tester = fault.Tester(sram_magma(family=family.MagmaFamily())())
  File "/nobackup/skavya/magma/magma/circuit.py", line 271, in __call__
    self = super(CircuitKind, cls).__call__(*largs, **kwargs)
  File "/nobackup/skavya/magma/magma/circuit.py", line 541, in __init__
    raise Exception("Can not instance a circuit outside a definition")
Exception: Can not instance a circuit outside a definition

leonardt, 2020-07-01:
Also feel free to post wherever :) if you have usage questions, posting them on GH issues is useful because others can google their error messages and perhaps the resolution here will help them

leonardt, 2020-07-01:
hmm that's an odd error, let me review the code, it's this module correct https://github.com/StanfordAHA/lake/blob/accessor/lake/modules/hwtypes/sram_stub.py?

kavyasreedhar, 2020-07-01:
Sounds great, thanks! I just pushed, so that file has the updated version now.

leonardt, 2020-07-01:
Ah, I believe there might an extra parenthesis pair in your Tester instantiation, you should be passing a circuit definition rather than an instance, so instead of 
```
tester = fault.Tester(sram_magma(family=family.MagmaFamily())())
```

I believe this should work
```
tester = fault.Tester(sram_magma(family=family.MagmaFamily()))
```

Basically, the extra parens are instancing the generated circuit definition, and it's complaining that you're trying to instance a circuit outside of an enclosing circuit definition.

kavyasreedhar, 2020-07-01:
That makes sense, thanks! I think I might be running into a set/get_item issue now because it seems like the selection range is inclusive instead of exclusive (so for 0:2, 2 is included instead of excluded). The error itself is just

ERROR: Cannot select 0:2 From self.addr
  Type: Bit[2]

so I wanted to check about that? I have also pasted the full error below in case it's helpful.

(hwtypes) skavya@kiwi:/nobackup/skavya/lake/lake/modules/hwtypes$ !p
python sram_stub.py
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
ERROR: Cannot select 0:2 From self.addr
  Type: Bit[2]

/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/coreir/libcoreir-c.so(_ZN6CoreIR8Wireable3selERKSs+0xc9)[0x7efc1e813a3d]
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/coreir/libcoreir-c.so(_ZN6CoreIR9ModuleDef3selERKSt5dequeISsSaISsEE+0xb3)[0x7efc1e7c64b3]
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/coreir/libcoreir-c.so(_ZN6CoreIR9ModuleDef3selERKSs+0x8a)[0x7efc1e7c60f4]
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/site-packages/coreir/libcoreir-c.so(COREModuleDefSelect+0x52)[0x7efc1e717bd2]
/usr/lib/x86_64-linux-gnu/libffi.so.6(ffi_call_unix64+0x4c)[0x7efc233d0dae]
/usr/lib/x86_64-linux-gnu/libffi.so.6(ffi_call+0x22f)[0x7efc233d071f]
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/lib-dynload/_ctypes.cpython-37m-x86_64-linux-gnu.so(_ctypes_callproc+0x2b4)[0x7efc235e4944]
/nobackup/skavya/python_envs/hwtypes/lib/python3.7/lib-dynload/_ctypes.cpython-37m-x86_64-linux-gnu.so(+0x120d4)[0x7efc235e50d4]
python(_PyObject_FastCallKeywords+0x4eb)[0x5d96db]
python[0x54aa51]
python(_PyEval_EvalFrameDefault+0x43f8)[0x551c08]
python(_PyFunction_FastCallKeywords+0x18c)[0x5d88dc]
python(_PyEval_EvalFrameDefault+0x68e)[0x54de9e]
python(_PyFunction_FastCallKeywords+0x18c)[0x5d88dc]
python(_PyEval_EvalFrameDefault+0x68e)[0x54de9e]
python(_PyFunction_FastCallKeywords+0x18c)[0x5d88dc]
python(_PyEval_EvalFrameDefault+0x68e)[0x54de9e]
python(_PyFunction_FastCallKeywords+0x18c)[0x5d88dc]
python(_PyEval_EvalFrameDefault+0x68e)[0x54de9e]
python(_PyFunction_FastCallKeywords+0x18c)[0x5d88dc]

leonardt, 2020-07-01:
That looks like it might be a compatibility issue with an older coreir version, can you trying upgrading with `pip install --upgrade coreir` and see if that helps?

kavyasreedhar, 2020-07-01:
Awesome, that works, thanks, I can see the waveforms!

One last thing - I had errors with magma_output="verilog" in tester.compile_and_run but could run without that argument. This does produce a verilog file in tempdir and it seems to have my module in it, so is this the verilog produced by that magma code?

Is it automatically generated now and do we not need to specify that magma_output argument?

leonardt, 2020-07-01:
Yes, the default magma compiler will generate verilog via coreir.  Using magma_output="verilog" will actually use the old verilog backend (non-corer) to produce verilog which is not actively maintained, so I would avoid using that unless required (I think there are some old use cases for doing this if you're importing external verilog, but since you're using magma code, I would just use the default compiler output)

kavyasreedhar, 2020-07-01:
Okay awesome, thank you!

leonardt, 2020-07-01:
I will close this issue for now, feel free to reopen if there's any problems that come up, or open a new one.

closed 2020-07-01 by leonardt